aurora-opensource/streetscape.gl #256: Select a vehicle and switch Camera Focus to it

tkt028, 2019-02-27:
**Context**
- in my project, I have a point cloud data (collection of points with coordinates and colors) and a JSON file describing the trajectory of vehicles (coordinates, orientation associated with timestamp).
- All the vehicles are equally important, there is no primary one. The coordinates are in absolute values.

**Requirement**
- I want to be able to dynamically select a vehicle (by mouse) to switch the Camera Focus so that I can have 3 views: Top-view, perspective and driver view of that vehicle.

**The current gap**
- I learned that XVIZ data (at least in the KITTI sample data) are organized around a primary vehicle (in the stream **/vehicle_pose**) that stream data from sensors can have a pose **offset relative** to that vehicle pose.

**Wishlist**
- data level (XVIZ): allows to represent coordinates in the absolute value, not offset relative to any primary vehicle.
- code level (streetscape.gl)
   - provide options to dynamically select a vehicle to switch the Camera focus to it.
   - allow to use the absolute value coordinates

xintongxia, 2019-02-27:
> **Wishlist**
>
> * data level (XVIZ): allows to represent coordinates in the absolute value, not offset relative to any primary vehicle.

> code level (streetscape.gl)
> ...
> * allow to use the absolute value coordinates

Our framework support different kinds of coordinate systems, you can find in xviz [doc](https://avs.auto/#/xviz/api-reference/xviz-builder/xviz-metadata-builder?section=methods) and search for `coordinate`. In your case, you can use `IDENTITY`.

Also there is another example [NuTonomy converter](https://github.com/uber/xviz/blob/master/examples/converters/nutonomy/src/converters/gps-converter.js#L84) using `IDENTITY` coordinate system.

But our framework do expect a `PRIMARY_VEHICLE_POSE` stream in every frame.

jlisee, 2019-02-27:
As mentioned by @xintongxia the default coordinate transform for a stream is `IDENTITY` so you can use absolute coordinates right now.  While we just made a vehicle pose optional #348, the camera controls are all based around the notion of there being a specific vehicle that you can care about.  The assumption was if you had no pose, all data was centered around the origin, like when viewing raw sensor data.

For now can you pick a vehicle manually while converting the data?  You can disable camera position and header tracking after the fact.

I agree it would be great to track a primitive instead of a vehicle, we have a similar feature in an internally streetscape.gl based application.

tkt028, 2019-03-07:
Thank you @jlisee and @xintongxia about the hints of coordinate topic. It's really helpful to my work.

**Technical note about this ticket:**
I dig into the code to find the solution for this feature. I learned that 
- (POINT 1) file streetscape.gl/modules/core/src/components/log-viewer/core-3d-viewer.js -> function _getViewState(): the information (longitude, latitude, bearing) of frame.trackPosition is used to calculate "trackedPosition" which will be passed to DeckGL for Camera settings -> I guess this point is the key to make change to solve this ticket by smartly building the right "trackedPosition" associated to the primary vehicle (pose) or a selected non-primary vehicle _(please help confirm this point)_.

- (POINT 2) I follow-up to see how "frame.trackPosition" is calculated. This question leads me to the file at xviz/modules/parser/src/parsers/parse-vehicle-pose.js where the calculation happen in the data conversion phase with XVIZ package.

```
export function getTransformsFromPose(vehiclePose) {
  const {longitude, latitude, altitude = 0} = vehiclePose;

  const origin =
    Number.isFinite(vehiclePose.longitude) && Number.isFinite(vehiclePose.latitude)
      ? [longitude, latitude, altitude]
      : null;
  const pose = new Pose(vehiclePose);

  const vehicleRelativeTransform = pose.getTransformationMatrix();

  // If mapOrigin is not specified, use a faux position of [0, 0, 0]
  // deck.gl needs a lon/lat position to target the camera
  const trackPosition = addMetersToLngLat(             // ####### trackPosition calculation happens here
    origin || [0, 0, 0],
    vehicleRelativeTransform.transformVector([0, 0, 0])
  );

  return {
    origin,
    vehicleRelativeTransform,
    trackPosition,
    heading: (pose.yaw / Math.PI) * 180
  };
}
```

At the root, this calculation requires below items
- class Pose for calculation (from https://github.com/uber-web/math.gl/blob/master/docs/api-reference/pose.md)
- with the inputs: location and orientation of primary vehicle (pose)

**MY HYPOTHESIS** (please help confirm if it's correct): by connecting the above 2 points, we have 2 below statements
- STATEMENT 1: in order to switch the Camera Focus to other non-primary vehicles (that respect to its moving orientation), we also need their orientation information to calcualte the Camera settings to pass to DeckGL in POINT 1
- STATEMENT 2: if STATEMENT 1 is correct, then it's XVIZ API limitation which currently doesn't support to include orientation information to vehicle primitives. See https://avs.auto/#/xviz/api-reference/xviz-builder/xviz-builder?section=XVIZPrimitiveBuilder

**QUESTION**: 
- Could you help confirm these points above?
- Do you have any concrete timeline to release this feature?

Thank you very much!

jlisee, 2019-03-07:
The vehicle based camera can track the vehicle but not track it's orientation.  Given that it should be possible for an initial version of this to be built without requiring any orientation information or control.

tkt028, 2019-05-22:
My workaround solution: 
- In the file [1], the function  _getViewState() calculates the parameters _trackedPosition_ (longitude, latitude, bearing) which tells the Camera where to focus. By making change in this entry point, I can resolve this problem.

[1] avs_visualizer/modules/core/src/components/log-viewer/core-3d-viewer.js

closed 2019-05-22 by tkt028

SlahserZ, 2021-12-27:
> My workaround solution:
>
> * In the file [1], the function  _getViewState() calculates the parameters _trackedPosition_ (longitude, latitude, bearing) which tells the Camera where to focus. By making [...]

could you give me some suggestions about this. I wanna control the camera target, but I just set **VIEW_MODE** but more detailed angle or pose. thank you.