aurora-opensource/streetscape.gl #257: Implementing your own streaming server

gerardsimons, 2019-02-27:
Hi,

I have my own robotics project going on which streams data over UDP. I wanted to implement some kind of middleware that converts these streams to XVIZ streams. I am having trouble figuring out the best approach for doing this.

On the one end, we have the examples in XVIZ, but they all seem to load files or generate frames (scenarios) beforehand and serving them in a rather complex way with xviz-serve-data.js. Not really what I am looking for.

Looking at streetscap.gl, I guess this [starter-kit section](https://github.com/uber/streetscape.gl/blob/master/docs/get-started/starter-kit.md#load-data-from-a-different-source) mentions what I would like to do, but the documentation is a bit limited. The client-side would as indicated be as easy as yarn start-streaming. The instructions on how to start a local XVIZ stream server just link back to the get-started docs which is not very helpful. 

I really love the play ground example and is kind of what I want to do, but then I would like to do the JSON part in a server that builds and sends some metadata and then builds new frames and sends them over to the streaming client based on some internal events. Am I missing something here and is it already there? 

Thanks again, love the work you guys are doing.

Pessimistress, 2019-02-27:
@gerardsimons Sorry that should have linked to [the XVIZ repo](https://github.com/uber/xviz/blob/master/docs/getting-started/running-the-example.md).

Streetscape currently offers implementations of loaders from two sources, file loader from static URL, and stream loader from WebSocket. If you need a different type of connection [this page](https://avs.auto/#/streetscape.gl/api-reference/loaders/xviz-loader-interface) might be helpful.

gerardsimons, 2019-02-27:
Thanks I did figure it out using a WebSocket server as you said. Still I think a very simple server would help out a lot of folks, especially those (like me) who are unfamiliar with NodeJS and WebSocket servers.

closed 2019-02-27 by gerardsimons

Pessimistress, 2019-02-27:
@gerardsimons The XVIZ repo offers the simple server in https://github.com/uber/xviz/tree/master/examples/server and we're planning to package it into a module when it's feature complete.

gerardsimons, 2019-02-28:
Indeed it does, but the example is quite complex and requires pre-loading the frames before serving. I think the easiest way to get started is actually a kind of streaming server that just sends some pose and other frames whenever it wants. I think the server in the examples is too complicated for many and might deter people from using it.

baker-Xie, 2019-04-02:
> Indeed it does, but the example is quite complex and requires pre-loading the frames before serving. I think the easiest way to get started is actually a kind of streaming server that just sends some pose and other frames whenever it wants. I think the server in the examples is too complicated for many and might deter people from getting into it

I'm really agree with you. The server example is very complex for me because I'm not familiar with nodejs. Meanwhile , the server example is realized in node.js. But in most autonomous application ,the data source application are written in c++/python. So, I think maybe we should send our data via TCP/UDP, and the nodejs server receive the data and convert it to XVIZ format in realtime, and then send the XVIZ data to streetscape? It's very complex...  BTW, I'm very appreciate with the work Uber are doing.