aurora-opensource/streetscape.gl #266: Broken link

johnwlambert, 2019-03-02:
On [this page](https://avs.auto/#/streetscape.gl/getting-started/starter-kit), I see a link to 
https://github.com/uber/streetscape.gl/tree/master/examples/get-started/log-from-file.js
However, this link is broken. 

Thanks for fixing the other link!

johnwlambert, 2019-03-02:
This link is also broken:

https://avs.auto/docs/api-reference/xviz-file-loader.md

along with this link:

https://avs.auto/docs/get-started/mapbox-tokens.md

jlisee, 2019-03-03:
Woops, we’ll port [this check](https://github.com/uber/xviz/commit/17ec0c67e08b202da5bbf50a019bc09f39b27c6b) from the XVIZ repo to save you the trouble from finding these the hard way.

Rastalamm, 2019-06-19:
@johnwlambert - Can you clarify which page(s) the below two links are referenced from? 

https://avs.auto/docs/api-reference/xviz-file-loader.md
https://avs.auto/docs/get-started/mapbox-tokens.md

As for the original broken link issue - I just created [a PR](https://github.com/uber/streetscape.gl/pull/351) to fix that.