aurora-opensource/streetscape.gl #273: Error when doing npm install streetscape.gl

shunjilin, 2019-03-11:
There seems to be an error when running npm install streetscape.gl, in particular when dealing with loaders.gl/gltf dependency
`
npm ERR! path /home/user/app/node_modules/@loaders.gl/gltf/bin/glbdump.js
npm ERR! code ENOENT
npm ERR! errno -2
npm ERR! syscall chmod
npm ERR! enoent ENOENT: no such file or directory, chmod '/home/user/app/node_modules/@loaders.gl/gltf/bin/glbdump.js'
npm ERR! enoent This is related to npm not being able to find a file.
npm ERR! enoent 

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/user/.npm/_logs/2019-03-11T08_54_26_434Z-debug.log
`

ibgreen, 2019-03-11:
This was fixed in a loaders.gl patch. I was under the impression that @twojtasz had updated streetscape to include that fix.

yiqin, 2019-03-12:
@ibgreen What is workaround to this issue?

gerardsimons, 2019-03-14:
Otherwise try `yarn add` rather than `npm install`

rob137, 2019-03-27:
`yarn add` worked for me, but `npm install` didn't.

twojtasz, 2019-03-28:
This was updated here: https://github.com/uber/xviz/pull/393 but it may not be published to npm.

I"ll get it published.

twojtasz, 2019-03-28:
ok, latest @xviz/parser@1.0.0-beta.6 addresses this issue.

closed 2019-03-28 by twojtasz