aurora-opensource/streetscape.gl #274: Polyline rendering / rounding issue

gerardsimons, 2019-03-12:
Hi everyone,

I am trying to draw a polyline with GEOGRAPHIC coordinates. I am using the `yarn start-live-local` from getting started. Before I was having this issue, so I am not sure if it's somehow related: https://github.com/uber/xviz/issues/402 

For testing I am just trying to draw a pretty huge line with just 2 vertices. Sometimes I do not see the polyline appear however, I seem to have narrowed down the problem to this:

```js
xvizMetadataBuilder
      .stream('/vehicle_pose')
      .category('pose')
      .stream("/ais/history")
      .category('primitive')
      .type('polyline')
      .coordinate('GEOGRAPHIC')
      .streamStyle({
        stroke_color: '#00FF00',
        stroke_width: 10,
        stroke_width_min_pixels: 1
      })

// Should draw a line intersecting your own verhicle
var polyLine = [[4.4186, 51.8889, 0], [4.5186, 51.8889, 0]];

// This does not show up anywhere
// var polyLine = [[4.4186, 51.8889, 0], [4.518599999999999, 51.8889, 0]]
xvizBuilder.pose('/vehicle_pose')
        .timestamp(timestamp)
        .mapOrigin(4.4186, 51.8889, 0)
        .position(0, 0, 0)
        .orientation(0, 0, 3);

xvizBuilder.primitive('/ais/history')
        .polyline(polyLine)
        .id("history")

gerardsimons, 2019-03-12:
I changed the name because I have a feeling it's not due to rounding but rather to an issue when the distance becomes too small. If I try this builder in the playground:

```
const timestamp = 1003.2;

  var lon =  4.4186;
  var lat = 51.8889;
  
  var aisOriginLon =  4.403540;
  var aisOriginLat = 51.891934;
  
  // https://www.google.nl/maps/place/51%C2%B047'26.6%22N+4%C2%B052'53.1%22E/@51.7911724,4.8712624,13.97z/data=!4m6!3m5!1s0x0:0x0!7e2!8m2!3d51.790732!4d4.8814194
  var wkLon = 4.881419;
  var wkLat = 51.790732;
  
  function interpolate(lon0, lat0, lon1, lat1, i) {
      var lon = lon0 + i * (lon1 - lon0);
      var lat = lat0 + i * (lat1 - lat0);
      
      return [lon, lat, 1]; // Alt is not interpolate because I don't really care
  }
  
  var aisLon = 4.881419;
  var aisLat = 51.790732;
  
  var start = [aisOriginLon, aisOriginLat, 1];
  var end = interpolate(aisOriginLon, aisOriginLat, wkLon, wkLat, 0.2);
    
  xvizBuilder.pose('/vehicle_pose')
      .timestamp(timestamp)
      .mapOrigin(lon, lat, 0)
      .orientation(0, 0, 0);
      
  xvizBuilder.primitive('/ais/history')
      .polyline([start,end])
      .id("history");
  
```

It does *not* work. If i change the interpolation to 0.3 it *does* work. Any thoughts on this?

gerardsimons, 2019-03-12:
The plot thickens, if I remove the altitude component it works. So actually if I give the other end of the polyline a different altitude it will work, at least that's what it looks like currently:

```js
function interpolate(lon0, lat0, lon1, lat1, i) {
      var lon = lon0 + i * (lon1 - lon0);
      var lat = lat0 + i * (lat1 - lat0);
      
      return [lon, lat, 0]; // As long as alt != 1 in my case
  }
```

Pessimistress, 2019-03-12:
This is due to https://github.com/uber/xviz/blob/master/modules/parser/src/config/xviz-config.js#L24 which is used to simplify paths. The default is 0.1 (meter) but in your case it is used on coordinates in degrees.

@twojtasz I think two things need to happen:

- The option should be documented
- The default should be 0

tkt028, 2019-03-13:
I have a similar issue of Polygon, I followed the hint of setting `pathDistanceThreshold `  to 0, but it doesn't work. Please see 
https://github.com/uber/streetscape.gl/issues/275#issuecomment-472301267

twojtasz, 2019-03-13:
Ok,  i'll add docs/default/tests