aurora-opensource/streetscape.gl #275: Polygon rendering issue in GEOGRAPHIC coordinate

tkt028, 2019-03-13:
Hi everyone,
I am working on drawing the Bounding Box for primitive object (polygon) in the coordinate GEOGRAPHIC.
The current situation: the Car icon is drawn in the right position, however, the Bounding Box polygon is not displayed.
I have tried different below experiments, but they don't work
- keep or remove the altitude component of Polygons points -> nothing happen
- provide 4, 5, 8 points of Bounding Box -> nothing happen

[image]

Sample XVIZ metadata
```
{"type":"xviz/metadata","data":{"version":"2.0.0","streams":{"/vehicle_pose":{"category":"pose"},"/tracklets/objects":{"category":"primitive","coordinate":"GEOGRAPHIC","stream_style":{"extruded":true,"fill_color":"#00000080"},"style_classes":[{"name":"Car","style":{"fill_color":"#50B3FF80","stroke_color":"#50B3FF"}},{"name":"Cyclist","style":{"fill_color":"#957FCE80","stroke_color":"#957FCE"}},{"name":"Pedestrian","style":{"fill_color":"#FFC6AF80","stroke_color":"#FFC6AF"}},{"name":"Van","style":{"fill_color":"#5B91F480","stroke_color":"#5B91F4"}},{"name":"Unknown","style":{"fill_color":"#E2E2E280","stroke_color":"#E2E2E2"}}],"primitive_type":"polygon"},"/tracklets/tracking_point":{"category":"primitive","stream_style":{"radius":0.2,"stroke_width":0,"fill_color":"#FFC043"},"coordinate":"GEOGRAPHIC","primitive_type":"circle"},"/tracklets/label":{"category":"primitive","stream_style":{"text_size":18,"fill_color":"#DCDCCD"},"coordinate":"GEOGRAPHIC","primitive_type":"text"},"/tracklets/trajectory":{"category":"primitive","stream_style":{"stroke_color":"#FFC043","stroke_width":0.1,"stroke_width_min_pixels":1},"coordinate":"GEOGRAPHIC","primitive_type":"polyline"}},"log_info":{"start_time":1534883786.267,"end_time":1534883906.7870002}}}
```

Sample XVIZ frame (10-frame.json)
```
{"type":"xviz/state_update","data":{"update_type":"snapshot","updates":[{"timestamp":1534883787.068,"poses":{"/vehicle_pose":{"timestamp":1534883787.068,"mapOrigin":{"longitude":-121.9318621943,"latitude":37.3146625269,"altitude":12.1044184975},"position":[0,0,0],"orientation":[0.786244846,-0.5200232285,-0.0794160647]}},"primitives":{"/tracklets/objects":{"polygons":[{"vertices":[[-121.9318335129,37.3146693562,0],[-121.9318484293,37.3146837662,0],[-121.9318762094,37.3146411597,0],[-121.9318911258,37.3146555697,0]],"base":{"object_id":"5c53aa359dd4e4448703f3d1","style":{"height":3.2244653986,"fill_color":"#50B3FF80","stroke_color":"#50B3FF","stroke_width":10},"classes":["Car"]}},{"vertices":[[-121.9318478464,37.3148138213,0],[-121.9318714361,37.3148134914,0],[-121.9318468267,37.3147673454,0],[-121.9318704164,37.3147670156,0]],"base":{"object_id":"5c53aa359dd4e4448703f3d2","style":{"height":3.386851358,"fill_color":"#50B3FF80","stroke_color":"#50B3FF","stroke_width":10},"classes":["Car"]}},{"vertices":[[-121.9320412355,37.3145428833,0],[-121.9320212437,37.3145518313,0],[-121.9320135459,37.3145033463,0],[-121.9319935541,37.3145122943,0]],"base":{"object_id":"5c53aa359dd4e4448703f3d3","style":{"height":3.3305248881,"fill_color":"#50B3FF80","stroke_color":"#50B3FF","stroke_width":10},"classes":["Car"]}},{"vertices":[[-121.9317004478,37.3143710638,0],[-121.9316995642,37.3143298072,0],[-121.9317214327,37.3143707806,0],[-121.9317205491,37.314329524,0]],"base":{"object_id":"5c53aa359dd4e4448703f3d4","style":{"height":3.0064703192,"fill_color":"#50B3FF80","stroke_color":"#50B3FF","stroke_width":10},"classes":["Car"]}},{"vertices":[[-121.9319522237,37.3148501747,0],[-121.9319512161,37.3148054013,0],[-121.9319296348,37.3148505096,0],[-121.9319286272,37.3148057361,0]],"base":{"object_id":"5c53aa369dd4e4448703f3d5","style":{"height":3.2482692813,"fill_color":"#50B3FF80","stroke_color":"#50B3FF","stroke_width":10},"classes":["Car"]}},{"vertices":[[-121.9320240482,37.3147395397,0],[-121.9320085619,37.3147784171,0],[-121.9320043784,37.3147345313,0],[-121.9319888921,37.3147734087,0]],"base":{"object_id":"5c53aa369dd4e4448703f3d6","style":{"height":3.0121022654,"fill_color":"#50B3FF80","stroke_color":"#50B3FF","stroke_width":10},"classes":["Car"]}},{"vertices":[[-121.9319117582,37.3147942844,0],[-121.9319130248,37.3147511925,0],[-121.9318899077,37.3147938764,0],[-121.9318911744,37.3147507846,0]],"base":{"object_id":"5c53aa369dd4e4448703f3d7","style":{"height":3.1406243451,"fill_color":"#50B3FF80","stroke_color":"#50B3FF","stroke_width":10},"classes":["Car"]}},{"vertices":[[-121.9319171455,37.3148854316,0],[-121.9318951268,37.3148859281,0],[-121.9319156177,37.3148420242,0],[-121.931893599,37.3148425206,0]],"base":{"object_id":"5c53aa369dd4e4448703f3d8","style":{"height":3.1640317346,"fill_color":"#50B3FF80","stroke_color":"#50B3FF","stroke_width":10},"classes":["Car"]}}]}}}]}}
```

tkt028, 2019-03-13:
I think this ticket somehow related to #274.

tkt028, 2019-03-13:
I followed the hint [1] to set XVIZ Config "pathDistanceThreshold" to 0, but it doesn't work too. Please see the screenshot.

[1] https://github.com/uber/streetscape.gl/issues/274#issuecomment-472080610

[image]

Pessimistress, 2019-03-13:
The error should have been fixed by https://github.com/uber/streetscape.gl/pull/272. Can you make sure you have the latest master?

tkt028, 2019-03-14:
Hi @Pessimistress ,
Thank you for the hints. You are right. By using the latest master, the Polygon is rendered now.
However, it seems that the Car icons are rendered in the right location while the polygons are rendered in the wrong place. From the below screenshot, it seems there is a constant distance between the Car icons and the Polygons.

Do you have any technical suggestion of Streetscapes.gl, xviz where I can take a look to troubleshoot this issue? Thank you very much!

[image]

Pessimistress, 2019-03-14:
First impression by looking at your sample data - your vehicle is at altitude `12.1044184975` (`[0, 0, 0]` meters from `mapOrigin`) and your objects are at altitude `0` (using absolute coordinates due to `GEOGRAPHIC` coordinate system). The 12m vertical difference is significant.

Pessimistress, 2019-03-14:
@tkt028 we have an experimental demo app at https://avs.auto/playground/index.html where you can live edit log metadata and frames.

tkt028, 2019-03-15:
Hi @Pessimistress ,
You are right. By setting the altitude to 0, the vehicle position is aligned with the MapBox.
I close this ticket now. Thank you very much!

closed 2019-03-15 by tkt028