aurora-opensource/streetscape.gl #277: Point cloud rendering issues in GEOGRAPHIC coordinate

tkt028, 2019-03-14:
Hi everyone,
I am adding a static point cloud (about surround environment, building) into XVIZ frames with coordinate GEOGRAPHIC. But the point cloud is not rendered (or at least I don't see it). NOTE: I have used the latest master code as suggested by [1].

[1] https://github.com/uber/streetscape.gl/issues/275#issuecomment-472749511

Sample XVIZ metadata
```
{"type":"xviz/metadata","data":{"version":"2.0.0","streams":{"/vehicle_pose":{"category":"pose"},"/tracklets/objects":{"category":"primitive","coordinate":"GEOGRAPHIC","stream_style":{"extruded":true,"fill_color":"#00000080"},"style_classes":[{"name":"Car","style":{"fill_color":"#50B3FF80","stroke_color":"#50B3FF"}},{"name":"Cyclist","style":{"fill_color":"#957FCE80","stroke_color":"#957FCE"}},{"name":"Pedestrian","style":{"fill_color":"#FFC6AF80","stroke_color":"#FFC6AF"}},{"name":"Van","style":{"fill_color":"#5B91F480","stroke_color":"#5B91F4"}},{"name":"Unknown","style":{"fill_color":"#E2E2E280","stroke_color":"#E2E2E2"}}],"primitive_type":"polygon"},"/tracklets/tracking_point":{"category":"primitive","stream_style":{"radius":0.2,"stroke_width":0,"fill_color":"#FFC043"},"coordinate":"GEOGRAPHIC","primitive_type":"circle"},"/tracklets/label":{"category":"primitive","stream_style":{"text_size":18,"fill_color":"#DCDCCD"},"coordinate":"GEOGRAPHIC","primitive_type":"text"},"/tracklets/trajectory":{"category":"primitive","stream_style":{"stroke_color":"#FFC043","stroke_width":0.1,"stroke_width_min_pixels":1},"coordinate":"GEOGRAPHIC","primitive_type":"polyline"},"/tracklets/pcd":{"category":"primitive","stream_style":{"fill_color":"#00a","radius_pixels":1},"coordinate":"GEOGRAPHIC","primitive_type":"point"}},"log_info":{"start_time":1534883786.267,"end_time":1534883906.7870002}}}
```

Sample XVIZ data (10-frame.json)

```
{"type":"xviz/state_update","data":{"update_type":"snapshot","updates":[{"timestamp":1534883787.068,"poses":{"/vehicle_pose":{"timestamp":1534883787.068,"mapOrigin":{"longitude":-121.9318621943,"latitude":37.3146625269,"altitude":12.1044184975},"position":[0,0,0],"orientation":[0.786244846,-0.5200232285,-0.0794160647]}},"primitives":{"/tracklets/objects":{"polygons":[{"vertices":[[-121.9318911258,37.3146555697,11.3402691875],[-121.9318762094,37.3146411597,11.3361635925],[-121.9318484293,37.3146837662,11.2606567824],[-121.9318335129,37.3146693562,11.2565511838]],"base":{"object_id":"5c53aa359dd4e4448703f3d1","style":{"height":1.6122326993,"fill_color":"#50B3FF80","stroke_color":"#50B3FF","stroke_width":10},"classes":["Car"]}},{"vertices":[[-121.9318714361,37.3148134914,11.1907580504],[-121.9318704164,37.3147670156,11.2288408587],[-121.9318478464,37.3148138213,11.1863402147],[-121.9318468267,37.3147673454,11.2244230229]],"base":{"object_id":"5c53aa359dd4e4448703f3d2","style":{"height":1.693425679,"fill_color":"#50B3FF80","stroke_color":"#50B3FF","stroke_width":10},"classes":["Car"]}},{"vertices":[[-121.9320412355,37.3145428833,10.8763067666],[-121.9320212437,37.3145518313,10.9462096337],[-121.9320135459,37.3145033463,10.9884449784],[-121.9319935541,37.3145122943,11.0583478436]],"base":{"object_id":"5c53aa359dd4e4448703f3d3","style":{"height":1.665262444,"fill_color":"#50B3FF80","stroke_color":"#50B3FF","stroke_width":10},"classes":["Car"]}},{"vertices":[[-121.9317214327,37.3143707806,11.2817443106],[-121.9317205491,37.314329524,11.3070090404],[-121.9317004478,37.3143710638,11.2109508188],[-121.9316995642,37.3143298072,11.2362155486]],"base":{"object_id":"5c53aa359dd4e4448703f3d4","style":{"height":1.5032351596,"fill_color":"#50B3FF80","stroke_color":"#50B3FF","stroke_width":10},"classes":["Car"]}},{"vertices":[[-121.9319522237,37.3148501747,10.959727142],[-121.9319512161,37.3148054013,11.0323894219],[-121.9319296348,37.3148505096,11.0401221234],[-121.9319286272,37.3148057361,11.1127844015]],"base":{"object_id":"5c53aa369dd4e4448703f3d5","style":{"height":1.6241346406,"fill_color":"#50B3FF80","stroke_color":"#50B3FF","stroke_width":10},"classes":["Car"]}},{"vertices":[[-121.9320240482,37.3147395397,10.7665560097],[-121.9320085619,37.3147784171,10.8053280981],[-121.9320043784,37.3147345313,10.8920508558],[-121.9319888921,37.3147734087,10.9308229415]],"base":{"object_id":"5c53aa369dd4e4448703f3d6","style":{"height":1.5060511327,"fill_color":"#50B3FF80","stroke_color":"#50B3FF","stroke_width":10},"classes":["Car"]}},{"vertices":[[-121.9319130248,37.3147511925,11.1973295743],[-121.9319117582,37.3147942844,11.1678109132],[-121.9318911744,37.3147507846,11.2289960459],[-121.9318899077,37.3147938764,11.1994773857]],"base":{"object_id":"5c53aa369dd4e4448703f3d7","style":{"height":1.5703121725,"fill_color":"#50B3FF80","stroke_color":"#50B3FF","stroke_width":10},"classes":["Car"]}},{"vertices":[[-121.9319171455,37.3148854316,11.0960450675],[-121.9319156177,37.3148420242,11.1314479643],[-121.9318951268,37.3148859281,11.1269183699],[-121.931893599,37.3148425206,11.1623212658]],"base":{"object_id":"5c53aa369dd4e4448703f3d8","style":{"height":1.5820158673,"fill_color":"#50B3FF80","stroke_color":"#50B3FF","stroke_width":10},"classes":["Car"]}}]},"/tracklets/pcd":{"points":[{"colors":[[7936,7936,7936],[7424,7424,7424]] ,"points":[[-121.932604845,37.3140647855,13.2691792464],[-121.9325815758,37.3140636102,13.2423595926]]}]}}}]}}
```

Pessimistress, 2019-03-14:
This is indeed something we do not support... May I ask why you want to use GEOGRAPHIC coordinates for point cloud? To accurately render lng/lat positions we need 64bit float for each point, comparing to 32bit when you use IDENTITY or VEHICLE_RELATIVE (i.e. meters).

tkt028, 2019-05-02:
**My solution**: I simply add DeckGL#PointCloudLayer to AVS and it works. I close this ticket now.

closed 2019-05-02 by tkt028