aurora-opensource/streetscape.gl #282: how to change color theme

pgh41, 2019-03-20:
In the docs, default theme is dark but i install it it was light-theme
how to change to the dark theme?
i want to know the file path that set the theme in /example/get-started

closed 2019-03-22 by pgh41

YunjieYu, 2019-05-30:
hi, how do you change the color theme?

sivakanth1989, 2019-06-13:
I added a toggle button and based on change, am switching between dark and light theme by maintaining the theme URL in the state like below.

`constructor(){
    super();
    this.state = {
      mapStyle: '',
      settings: {
        switchTheme: false
      }
    }
  }
_onSettingsChange = changedSettings => {
    if(changedSettings.switchTheme){
      this._Dark();
    } else {
      this._Light();
    }
  };

  _Dark = () => {
    this.setState({
      mapStyle: 'mapbox://styles/mapbox/dark-v10',
      settings: {switchTheme: true}
    });
  }

  _Light = () => {
    this.setState({
      mapStyle: 'mapbox://styles/mapbox/light-v10',
      settings: {switchTheme: false}
    });
  }`
And passing the 'mapStyle' as props to 'LogViewer'
`render (){
    return (
      <ThemeProvider>
        <div>
         <Form
            data={FORM_FIELDS_READ_FROM_CONSTANTS_FILE}
            values={this.state.settings}
            onChange={this._onSettingsChange}              
          />
        </div>
        <LogViewer mapStyle={this.state.mapStyle} />
      </ThemeProvider>
    );
  }`

FORM_FIELDS_READ_FROM_CONSTANTS_FILE will be a toggle button defined in constants file.

Am not sure this is the right way or not. But this is working for me.
Thank you.