aurora-opensource/streetscape.gl #299: Video Flicker when streaming image blobs

jruddell, 2019-04-18:
When running from live, each frame is passed as an image blob. After digging into the code I noticed the [image-sequence.js](https://github.com/uber/streetscape.gl/blob/master/modules/core/src/components/declarative-ui/image-sequence.js#L108) class never has the image data loaded (because the browser needs to load it). </br>
So essentially what happens is `componentWillReceiveProps` sets state from `_getCurrentFrames `. This kicks off the image load promise, but returns `currentFrameImage` as undefined. The image subsequently is removed, then the promise callback fires and sets the new frame into state.

The fix for this would be to not remove the image but rather just not change the `currentFrameImage` if it hasn't loaded yet.

```
  _getCurrentFrames(props) {
    const {currentTime, src} = props;
    const {currentFrameImage} = this.state || {}

    const currentFrame = this._buffer.set(src, currentTime);
    const currentFrameData = this._buffer.get(currentFrame);

    if (currentFrameData && !currentFrameData.image) {
      currentFrameData.promise.then(image => {
        if (this.state.currentFrame === currentFrame) {
          this.setState({currentFrameImage: image});
        }
      });
    }

    return {
      currentFrameImage: (currentFrameData && currentFrameData.image) || currentFrameImage,
      currentFrame
    };
  }
```

I'm happy to make a PR if this is unintended behavior / a bug. If its not can you explain how I should stream image blobs from live data?

Algirdyz, 2019-04-25:
I have found the same issue as well.

I personally fixed it by removing the line here:
https://github.com/uber/streetscape.gl/blob/7313aec9c35b8c4d2bde7eea4490b7e5c6c27566/modules/core/src/components/declarative-ui/image-sequence.js#L148

Which clears the image if there is no new one.

MGraefe, 2019-05-06:
This should be fixed now: https://github.com/uber/streetscape.gl/pull/310

jruddell, 2019-05-06:
@MGraefe Great thanks! When will this be released (so I can pull down in NPM)?

Pessimistress, 2019-05-07:
Fix is published in 1.0.0-beta.10.

closed 2019-05-09 by Pessimistress

jruddell, 2019-05-13:
Thanks, I was able to verify the updated code is pulled in!