aurora-opensource/streetscape.gl #307: Point rendering issue in GEOGRAPHIC coordinate

jack-wolfs, 2019-04-29:
hi everyone
I'm trying to draw a point with GEOGRAPHIC coordinates. but I never see the point appear. I'm not sure if it is a bug or something wrong in my codes.
`// metadata
xvizMetadataBuilder
  .stream('/vehicle_pose')
  .category('pose')

  .stream('/lidar/points')
  .category('primitive')
  .type('point')
  .coordinate('GEOGRAPHIC')
  .streamStyle({
    point_color_mode: 'elevation',
    point_color_domain: [0, 3],
    radius_pixels: 10
  })

xvizBuilder
  .pose('/vehicle_pose')
  .timestamp(timestamp)
  .mapOrigin(-122.4, 37.8, 0)
  .orientation(0, 0, 0)

xvizBuilder
  .primitive('/lidar/points')
  .points([-122.401,37.8,0])`

pkulijing, 2019-05-06:
Same issue here. It seems the problem has been solved in the newest version.

twojtasz, 2019-05-20:
@pkulijing thanks for the update that it works with newest version.  @jack-wolfs can you try again?

sivakanth1989, 2019-06-20:
Am able to see points in the new version. One thing here, lidar point color am able to give in the playground, but in the get-started application it's not working.