aurora-opensource/streetscape.gl #321: Cant visualize XViz generated data

MatthiasThDs, 2019-05-09:
Hello,
I get the following errors with a fresh install from today.
[image]

I installed XViz using get started tutorial:
download-kitti-data.sh
run-kitti-example.sh

After that i installed streetscrape.gl using get started tutorial

I copied the generated "2011_09_26_drive_0005_sync" files (0-frame.json and xxx-frame.glb) from xviz/data/generated to streetscape.gl\examples\get-started\ and changed the  examples/get-started/app.js to:

```
export default new XVIZFileLoader({
  timingsFilePath:
    './2011_09_26_drive_0005_sync/0-frame.json',
  getFilePath: index =>
    `./2011_09_26_drive_0005_sync/${index + 1}-frame.glb`,
  worker: true,
  maxConcurrency: 4
});
```

Whats wrong? :)
Best regards
Matthias

Pessimistress, 2019-05-09:
I just did a clean reinstall and it is working. Can you document the commands you used to install the dependencies?

MatthiasThDs, 2019-05-13:
Hi,

I realized that the problem might be that I generated the glb files on a linux machine and then copied them over to windows. On Windows then i get the error documented above. If i use the original glb files from the website (avs.auto) everything works fine.

I'll ask in the xviz project how to generate the files on windows.
Best regards
Matthias

twojtasz, 2019-06-05:
@MatthiasThDs  You should not have to copy the files over if you are using the instruction for the quick start.  The script `./scripts/run-kitti-example.sh` should run the conversion and then start the XVIZ server.

Then in **streetscape.gl** you do have to run `yarn start-streaming-local.
Running `yarn start` won't connect to the local server i believe which is why you probably copied them over.  I will make this clearer in the docs.

Also curious on the linux <-> windows and if that was a problem or not

sivakanth1989, 2019-06-13:
First, you have to signup in https://www.mapbox.com/
Then go https://account.mapbox.com/access-tokens/ and create a token.
Copy and add it in the constants.js file or directly pass to LogViewer like below.

`<LogViewer {other_props} mapboxApiAccessToken="generated_token" />`

Hope this will help you.
Thank you

VijeetBenni, 2019-09-13:
In contstants.js file where to add the access token?

VijeetBenni, 2019-09-13:
With out the mapbox, 

> First, you have to signup in https://www.mapbox.com/
> Then go https://account.mapbox.com/access-tokens/ and create a token.
> Copy and add it in the constants.js file or directly pass to LogViewer like [...]

In contstants.js file where to add the access token?

VijeetBenni, 2019-09-13:
> First, you have to signup in https://www.mapbox.com/
> Then go https://account.mapbox.com/access-tokens/ and create a token.
> Copy and add it in the constants.js file or directly pass to LogViewer like [...]

Even if Mapbox isnt available , streetscape should visualize the lidar points and the bounding boxes right?
I am not able to see them. But I am able to see acceleration, velocity, wheel values changing when streaming.

sivakanth1989, 2019-09-14:
WIthout map box token streetscape won't show map and lidar data.
You can find constants.js file in **/streetscape/examples/get-started/src/**

VijeetBenni, 2019-09-16:
> WIthout map box token streetscape won't show map and lidar data.
> You can find constants.js file in **/streetscape/examples/get-started/src/**

Thanks.

Does Kitti dataset  providing the geo coordinates to mapbox? 
I am running the kitti converter to convert kitti data into XVIZ format and using it display in streetscape.

For mapbox token I just set the environment variable instead of adding it in constants.js file, will it work?

VijeetBenni, 2019-09-16:
Tried providing token in constants.js, no luck yet.

VijeetBenni, 2019-09-16:
> Even if Mapbox isnt available , streetscape should visualize the lidar points and the bounding boxes right?
> I am not able to see them. But I am able to see acceleration, velocity, wheel values [...]

While creating the token, what all scopes need to select in secret?

afei26579, 2019-10-25:
> Hi,
>
> I realized that the problem might be that I generated the glb files on a linux machine and then copied them over to windows. On Windows then i get the error documented above. If i use the [...]

> Hi,
>
> I realized that the problem might be that I generated the glb files on a linux machine and then copied them over to windows. On Windows then i get the error documented above. If i use the [...]

i have the same question. it can not find local file . this is my config
[image]
and the bower show:
[image]

How to do ？

srikanth-kuriyala, 2020-03-17:
Anyone got the solution?

srikanth-kuriyala, 2020-03-18:
> Anyone got the solution?

I placed the "data" folder inside the public folder and I tried in the below way, It's working. 

import { resolve } from 'path';

const DATA_DIR = resolve(__dirname, "./data");

export default new XVIZFileLoader({
  timingsFilePath: `${DATA_DIR}/0-frame.json`,
  getFilePath: index =>`${DATA_DIR}/${index + 1}-frame.glb`,
  worker: true,
  maxConcurrency: 4
});

jayanshuagrawal, 2022-04-10:
I am also getting the same mapbox token issue. 
Can someone tell how we have to pass the token. I have followed the steps mentioned above and generated the token from mapbox website, But i am not understanding how to pass this token into constant.js mentioned above.

can someone help me