aurora-opensource/streetscape.gl #332: Can I run this client remotely on the server?

tower111, 2019-05-22:
Can I run this client remotely on the server, if I need to set it up?
thank you.

jlisee, 2019-05-23:
Since this is a web application you can server it from a remote server.  In that case what you have to take into consideration is browser security settings.  If you serve the application over HTTPS then your XVIZ server will need to have a SSL websocket connection, the requirement is dropped if you are using HTTP.

tower111, 2019-05-23:
> Since this is a web application you can server it from a remote server. In that case what you have to take into consideration is browser security settings. If you serve the application over HTTPS [...]

This is the case, my xviz server can provide data stream to streetscape.gl. I am now running streetscape.gl on the server, remotely only through the browser without having to run other programs. Can this be done? If you need to set it up

twojtasz, 2019-06-05:
The example applications are Single Page Applications, so you can host the files remotely but eventually you have to access the web application in a browser.

It is not clear to me what you want to achieve.  Do you want to literally run the web application on the server, or just have it hosted on a server but connect to it via a web browser?

tower111, 2019-06-09:
> The example applications are Single Page Applications, so you can host the files remotely but eventually you have to access the web application in a browser.
>
> It is not clear to me what you want to [...]

Sorry, my English is not very good.
I was trying to run on the server only by the browser.
already solved. It is a problem with my firewall. thanks

twojtasz, 2019-07-08:
Glad you solved it!

closed 2019-07-08 by twojtasz