aurora-opensource/streetscape.gl #334: when i run yarn build     Cannot find module '../aliases'

tower111, 2019-05-23:
[root@localhost website]# yarn
yarn install v1.15.2
[1/5] Validating package.json...
[2/5] Resolving packages...
success Already up-to-date.
Done in 0.74s.
[root@localhost website]# yarn build
yarn run v1.15.2
$ ./scripts/build.sh
[1/5] Validating package.json...
[2/5] Resolving packages...
success Already up-to-date.
/root/AutoDrive/streetscape.gl/website/node_modules/webpack-cli/bin/cli.js:74
				throw err;
				^

Error: Cannot find module '../aliases'
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:581:15)
    at Function.Module._load (internal/modules/cjs/loader.js:507:25)
    at Module.require (internal/modules/cjs/loader.js:637:17)
    at require (/root/AutoDrive/streetscape.gl/website/node_modules/v8-compile-cache/v8-compile-cache.js:161:20)
    at Object.<anonymous> (/root/AutoDrive/streetscape.gl/website/webpack.config.js:25:17)
    at Module._compile (/root/AutoDrive/streetscape.gl/website/node_modules/v8-compile-cache/v8-compile-cache.js:192:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:700:10)
    at Module.load (internal/modules/cjs/loader.js:599:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:538:12)
    at Function.Module._load (internal/modules/cjs/loader.js:530:3)
    at Module.require (internal/modules/cjs/loader.js:637:17)
    at require (/root/AutoDrive/streetscape.gl/website/node_modules/v8-compile-cache/v8-compile-cache.js:161:20)
    at WEBPACK_OPTIONS (/root/AutoDrive/streetscape.gl/website/node_modules/webpack-cli/bin/utils/convert-argv.js:115:13)
    at requireConfig (/root/AutoDrive/streetscape.gl/website/node_modules/webpack-cli/bin/utils/convert-argv.js:117:6)
    at /root/AutoDrive/streetscape.gl/website/node_modules/webpack-cli/bin/utils/convert-argv.js:124:17
    at Array.forEach (<anonymous>)
    at module.exports (/root/AutoDrive/streetscape.gl/website/node_modules/webpack-cli/bin/utils/convert-argv.js:122:15)
    at yargs.parse (/root/AutoDrive/streetscape.gl/website/node_modules/webpack-cli/bin/cli.js:71:45)
    at Object.parse (/root/AutoDrive/streetscape.gl/website/node_modules/yargs/yargs.js:567:18)
    at /root/AutoDrive/streetscape.gl/website/node_modules/webpack-cli/bin/cli.js:49:8
    at Object.<anonymous> (/root/AutoDrive/streetscape.gl/website/node_modules/webpack-cli/bin/cli.js:375:3)
    at Module._compile (internal/modules/cjs/loader.js:689:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:700:10)
    at Module.load (internal/modules/cjs/loader.js:599:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:538:12)
    at Function.Module._load (internal/modules/cjs/loader.js:530:3)
    at Module.require (internal/modules/cjs/loader.js:637:17)
    at require (internal/modules/cjs/helpers.js:20:18)
    at Object.<anonymous> (/root/AutoDrive/streetscape.gl/website/node_modules/webpack/bin/webpack.js:156:2)
    at Module._compile (internal/modules/cjs/loader.js:689:30)
error Command failed with exit code 1.
info Visit https://yarnpkg.com/en/docs/cli/run for documentation about this command.
[root@localhost website]# npm -v
6.4.1
[root@localhost website]# node -v
v10.10.0
[root@localhost website]# yarn -v
1.15.2

twojtasz, 2019-06-05:
@tower111 pull down the latest.  We have moved to a new module to manages what `aliases` was doing previously.

twojtasz, 2019-07-08:
The *ocular-dev-tools* module handles this now automatically that we no longer have to maintain a separate list. *master* is working.

closed 2019-07-08 by twojtasz