aurora-opensource/streetscape.gl #337: Mapbox token issue

walter-grace, 2019-05-29:
Hey there, I am having difficulty placing my Mapbox token to get access to the API, the instructions link in the get-started isn't working either.

I tried creating an  .env*.local file and created MAPBOX_TOKEN = _MYTOKEN_ then when to app.js ...
 <LogViewer mapboxApiAccessToken={process.env.MAPBOX_TOKEN}...
 but the process.env isn't reading; maybe something on my end? Thanks for the help

twojtasz, 2019-06-05:
I see the env var `const MAPBOX_TOKEN = process.env.MapboxAccessToken; // eslint-disable-line` in the code not **MAPBOX_TOKEN**

Could you try `export MapboxAccessToken="...."`

walter-grace, 2019-06-05:
Roger that, I actually just went into constants and added my token as a string, and replaced process.env. That worked for me.

closed 2019-06-05 by walter-grace