aurora-opensource/streetscape.gl #345: How to change lidar points color

sivakanth1989, 2019-06-13:
Am trying to change the color of lidar points in below way but it's not taking effect in the front end. All points color are showing in black. If I change 'point_color_mode' to 'elevation' then different colors are coming for points. I want to make all the points in the same color. Is it possible?

` const xviz_style = {
      '/tracklets/objects': [{
        name: 'highlighted',
        style: {fill_color: '#ff8000aa'}
      }],
      '/lidar/points': [{
        style: {
          point_color_mode: 'default',
          radius_pixels: settings.dotSize,
          fill_color: '#ff8000aa'
        }
      }]
    }`

I followed with the reference https://github.com/uber/xviz/blob/master/docs/protocol-schema/style-specification.md#point

Please suggest.
Thank you

twojtasz, 2019-06-21:
Can you try here: https://avs.auto/playground/

Setting the color in the *metadata* here works for me.  Setting the color on the object does not seem to work :(

```
// frame
const timestamp = 1000;

function rnd(max) {
    return Math.random() * max;
}

function randomPoints(n) {
    const points = new Array(n).fill([])
    points.forEach((_, i, arr) => arr[i] = [0, 0, 0].map(v => Math.random() * 10))
    return points;
}

xvizBuilder
  .pose('/vehicle_pose')
  .timestamp(timestamp)
  .mapOrigin(-122.4, 37.8, 0)
  .orientation(0, 0, 0)

xvizBuilder
  .primitive('/points')
  .points(randomPoints(100), 1);

```

```
// metadata
xvizMetadataBuilder
  .startTime(1000)
  .endTime(1005)

  .stream('/vehicle_pose')
  .category('pose')

  .stream('/points')
  .category('primitive')
  .type('points')
  .coordinate('VEHICLE_RELATIVE')
  .streamStyle({
    radius_pixels: 5,
    fill_color: [255, 0, 0]
  })
```

jlisee, 2019-07-12:
I was able to use `colors` with an array that matches the given size:

```
xvizBuilder
  .primitive('/points')
  .points(randomPoints(4), 1)
  .colors([[20, 100, 20],[20, 100, 20],[20, 100, 20],[20, 100, 20]])
```

According to the style spec point styles only apply when applied at the stream level not went sent with the object.  I think we should consider fixing this.

peppahjackk, 2021-09-16:
Is there a way to change the point cloud colors in the frontend after conversion?

I'm trying to use Deck.gl [PointCloudLayer](https://deck.gl/docs/api-reference/layers/point-cloud-layer) but the data format picked up from my converted pointcloud stream does not line up.