aurora-opensource/streetscape.gl #346: Add left side and right side view of vehicle

sivakanth1989, 2019-06-13:
Can anyone suggest, how to add left side view and right side view of a vehicle?

Thank you.

twojtasz, 2019-06-21:
It is not completely clear what you are trying to achieve.
We have some documentation how to display 2 logs.  This creates 2 viewports with different data sources but synchronized controls and camera.  https://avs.auto/#/streetscape.gl/advanced-topics/display-multiple-logs?section=synchronized-cameras

If you just want an additional 3d view from the same data, you should be able to create that using multiple LogViewers and updating the [viewState](https://avs.auto/#/streetscape.gl/api-reference/react-components/log-viewer?section=viewstate-object-optional-).  I'm not 100% sure on this but if you can access the view state of you main window then you should be able to adjust the other viewstates with a fixed transform (ie left & right)

twojtasz, 2019-07-08:
@sivakanth1989 Let me know if the above is useful or not.

sivakanth1989, 2019-07-12:
@twojtasz Thank you for the reply. 
My question is, right now we have 3 view modes: Perspective, top-down and Driver. Now I want to add Left view and Right view to see the vehicle from the left side and right side.

Could you please help me.

vinnnnnnu, 2021-04-10:
Hi @twojtasz and @srikanth-kuriyala, 

Even I am looking for implementing same kind of requirement.  Mainly I am trying to achieve rear view,  can you please help me out in this?  Thank you!

pai-qi, 2021-06-20:
> @twojtasz Thank you for the reply.
> My question is, right now we have 3 view modes: Perspective, top-down and Driver. Now I want to add Left view and Right view to see the vehicle from the left side and right side.
> 
> Could you please help me.

Hi, is it done yet?

vinnnnnnu, 2021-06-22:
Hello, nope.. still haven't done?
any suggestions?

SlahserZ, 2022-03-09:
That's easy. you can set your `car offset` bearing as 90 or -90, you can get it.