aurora-opensource/streetscape.gl #360: Rendering kepler.gl like heatmaps

gerardsimons, 2019-07-10:
Hi there,

We would like to plot something like a heatmap on our maps. Imagine something like a probability distribution rather than a discrete point in time. The beautiful examples shown at [kepler.gl](https://kepler.gl/) would be perfect. 

I talked about this at the Slack channel with @twojtasz where he mentioned this would be an interesting question / challenge. If I understand correctly kepler.gl uses deck.gl as well for rendering data on a map so there might be some overlap.

Any idea if this is already possible with some trickery or when a more official collaboration is planned including examples? Or would my best bet be to use something else like [custom layers ](https://avs.auto/#/streetscape.gl/advanced-topics/add-custom-3d-layers?section=render-custom-map-layers)?

Thanks again.

jlisee, 2019-07-12:
What is the data source you want to use to drive the heat maps?  Could you use an image (uber/xviz#415) displayed in a flat plane with alpha providing the transparency where there is no data?  Maybe with support for "raw" RGB images?

gerardsimons, 2019-07-12:
Yes that would already help a lot. The data actually comes from a predictive model that outputs probability distributions (gaussian mixture model). We could pre-rasterize it into an image and send that over for now.

gerardsimons, 2019-07-16:
Actually we decided to use the ColumnLayer with a point primitive which was quite easy to do as well. I can share a screenshot once it's pretty