aurora-opensource/streetscape.gl #362: layer stuck on awaiting state

iosefa, 2019-07-15:
Hello, 
I have an issue that seems to be related to #357, where I cannot seem to add custom layers. I have added created a point cloud layer as:

```
export default new PointCloudLayer({
    id: 'point-cloud-layer',
    data: {position: [0, 0, 0], normal: [0, 0, 1], color: [255, 255, 0]},
    pickable: false,
    coordinateSystem: COORDINATE_SYSTEM.METER_OFFSETS,
    coordinateOrigin: [0, 0],
    radiusPixels: 100,
    getPosition: d => d.position,
    getNormal: d => d.normal,
    getColor: d => d.color
});
```

and importing it in `app.js` , and adding it to customLayers as:

```
            <LogViewer
              log={log}
              mapboxApiAccessToken={MAPBOX_TOKEN}
              mapStyle={MAP_STYLE}
              car={CAR}
              xvizStyles={XVIZ_STYLE}
              viewMode={VIEW_MODE[settings.viewMode]}
              customLayers={[layer]}
            />
```

console.log of the layer reveals:

```
PointCloudLayer {props: {…}, id: "point-cloud-layer", count: 0, lifecycle: "Awaiting state", parent: null, …}
context: {layerManager: LayerManager, deck: Deck, gl: WebGL2RenderingContext, time: 250238, useDevicePixels: true, …}
count: 0
id: "point-cloud-layer"
internalState: LayerState {component: PointCloudLayer, asyncProps: {…}, onAsyncPropUpdated: ƒ, oldProps: {…}, oldAsyncProps: null, …}
lifecycle: "Matched. State transferred from previous layer"
parent: null
props: {id: "point-cloud-layer", pickable: false, coordinateSystem: 2, coordinateOrigin: Array(3), radiusPixels: 100, …}
state: {attributeManager: AttributeManager, model: Model}
stats: (...)
__proto__: Layer
```

If I am going about adding a point cloud in the wrong way, please advise. Thank you!

iosefa, 2019-07-15:
The funny thing is that this seemed to be working the day before yesterday...

OmarTag, 2020-06-28:
Any updates on this ?