aurora-opensource/streetscape.gl #363: draw results in the video stream

gouchaoer, 2019-07-15:
hi, I want to send live video stream or video file into the browser and draw some pictures on every frame to show the perception result. Of course the perception result is sent from the server.
How to implement this via uber's avs tools?

jlisee, 2019-07-16:
We don't have direct support for this right now at a high level what you want to render the 3D world from the perspective of the camera image you are sending, then draw the image behind that world so it overlays on top of it.

Concretely you would:
 - Send the images as XVIZ image stream
 - Send what you want to draw as XVIZ streams
 - Render the image to a full screen Quad
 - Adjust the camera parameters of deck.gl to match that of the camera your image came from

This is actually a long term planned feature but we have not implemented it yet.

pgebhardt, 2020-01-14:
Hi,

are there still plans for this feature to implemented anytime soon? I am actively using streetscape for visualization of our vehicle data, and this feature is highly anticipated.

I am looking forward to read from you guys soon.

soodla, 2020-05-06:
Hello!

Are there any updates on this feature?