aurora-opensource/streetscape.gl #370: Can streetscape.gl listen to mapbox's own event

Questionboy, 2019-07-24:
hi, 
I make an underground parking map by geoserver. And I load the map through change the  MAP_STYLE. I want to click the map and get the parking space information from map. 
I see that streetscape.gl use the react-map-gl to load map. In module of components/log-viewer/core-3d-viewer.js, the load event of StaticMap can get the map instance of mapbox. So I register mapbox's own event in here.  Just like this:

 **_onMapLoad = evt => {
    const map = evt.target;
    map.on('click', function (e) {
      console.log("mapbox click");
    });
    map.on('click', 'newpark', function (e) {
      console.log("mapbox layer click");
    });
    this.props.onMapLoad(map);
  };**

Howerver It doesn't work!
Is there any way to listen to mapbox's own event in streetscape.gl

twojtasz, 2019-08-12:
@Pessimistress any suggestions?

Pessimistress, 2019-08-12:
We disable pointer interaction on the map. You can do something like

```js
onClick={evt => {
  this._map.queryRenderedFeatures(...);
}}
_onMapLoad={evt => {
  this._map = evt.target;
}}
```

Questionboy, 2019-08-13:
Thank you for your reply. As you say, I change the StaticMap to InteractiveMap and try to add the onClick event in InteractiveMap. However, I cann't get any return from onClick event in InteractiveMap. It must be intercept by the click event in deck.gl.

JianXinyu, 2021-12-23:
> ```js
> queryRenderedFeatures
> ```

Hi, did you solve this in the end? I have the same problems now. Thanks!