aurora-opensource/streetscape.gl #373: Panel is undefined in xviz-panel.js

dks333, 2019-07-30:
Any advice?
[image]

dks333, 2019-07-30:
Local web page is blank due to this error.

gtangxiaobo, 2019-07-31:
same problem

leicao-me, 2019-08-01:
same problem

dks333, 2019-08-01:
In xviz-panel.js(path: streetscape.gl/modules/core/src/components/declarative-ui/), change the lines at the bottom to below. Then, run "yarn start-local"

`const getLogState = (log, ownProps) => {
  const metadata = log.getMetadata();
  const panel = metadata && metadata.ui_config && metadata.ui_config[ownProps.name];
  return {
-    uiConfig: panel.config || panel
+    uiConfig: panel && panel.config || panel
  };
}`

dks333, 2019-08-01:
In xviz-panel.js(path: streetscape.gl/modules/core/src/components/delarative-ui/), change the lines at the bottom to below, then run ""yarn start-local".

const getLogState = (log, ownProps) => {
  const metadata = log.getMetadata();
  const panel = metadata && metadata.ui_config && metadata.ui_config[ownProps.name];
  return {
    uiConfig: panel && panel.config || panel
  };
}

leicao-me, 2019-08-09:
I got the same problem, but when I used the release 1.0 Beta 5, it works: https://github.com/uber/streetscape.gl/releases/tag/v1.0.0-beta.5

I guess it's a bug for the current versions.