aurora-opensource/streetscape.gl #376: "get-started" error on virtual machine

user-Mrh, 2019-08-06:
hi guys,
I encountered the following error while running the demo:
[image]
[image]
Has anyone come across this problem or can someone give me some guidance?

ShanmukhaManoj11, 2019-08-09:
I encountered the same on a virtual machine. Did you find any solution for this? Does the problem occur only when run on a virtual machine?

leicao-me, 2019-08-09:
I got the same problem, but when I used the release 1.0 Beta 5, it works: https://github.com/uber/streetscape.gl/releases/tag/v1.0.0-beta.5

I guess it's a bug for the current versions.

twojtasz, 2019-08-12:
@user-Mrh could you try running the latest master with `yarn start-local` or `yarn start-streaming-local`?

There was a fix but I have not published the latest just yet, will do so today.

For references, `start-local` uses the local source code rather than the published packages.
`start-streaming-local` use the local source code but also will load XVIZ data from a local server @ port 8081 for the get-started demo rather than fetching data from S3.

I'll add this definitions to the README for the get started if it's not already there.

user-Mrh, 2019-08-13:
@twojtasz Thanks for your reply.

twojtasz, 2019-08-13:
@user-Mrh did those suggestions work?

user-Mrh, 2019-08-14:
> 
> 
> @user-Mrh did those suggestions work?

Yes, now I've downloaded the latest version of master and it works fine.

closed 2019-08-15 by user-Mrh