aurora-opensource/streetscape.gl #377: The system cannot find the path specified. error Command failed with exit code 1.

sandeepmeenuga, 2019-08-11:
$ yarn bootstrap                                                                                                                                                                                 yarn run v1.13.0                                                                                                                                                                                 $ yarn && ocular-bootstrap                                                                                                                                                                       warning package-lock.json found. Your project contains lock files generated by tools other than Yarn. It is advised not to mix package managers in order to avoid resolution inconsistencies cau sed by unsynchronized lock files. To clear this warning, remove package-lock.json.                                                                                                               [1/5] Validating package.json...                                                                                                                                                                 [2/5] Resolving packages...                                                                                                                                                                      [3/5] Fetching packages...                                                                                                                                                                       info fsevents@1.2.9: The platform "win32" is incompatible with this module.                                                                                                                      info "fsevents@1.2.9" is an optional dependency and failed compatibility check. Excluding it from installation.                                                                                  [4/5] Linking dependencies...                                                                                                                                                                    warning " > @babel/plugin-proposal-class-properties@7.4.4" has unmet peer dependency "@babel/core@^7.0.0-0".                                                                                     warning "@babel/plugin-proposal-class-properties > @babel/helper-create-class-features-plugin@7.4.4" has unmet peer dependency "@babel/core@^7.0.0".                                             warning " > @babel/preset-react@7.0.0" has unmet peer dependency "@babel/core@^7.0.0-0".                                                                                                         warning "@babel/preset-react > @babel/plugin-transform-react-display-name@7.2.0" has unmet peer dependency "@babel/core@^7.0.0-0".                                                               warning "@babel/preset-react > @babel/plugin-transform-react-jsx@7.3.0" has unmet peer dependency "@babel/core@^7.0.0-0".                                                                        warning "@babel/preset-react > @babel/plugin-transform-react-jsx-self@7.2.0" has unmet peer dependency "@babel/core@^7.0.0-0".                                                                   warning "@babel/preset-react > @babel/plugin-transform-react-jsx-source@7.2.0" has unmet peer dependency "@babel/core@^7.0.0-0".                                                                 warning "@babel/preset-react > @babel/plugin-transform-react-jsx > @babel/plugin-syntax-jsx@7.2.0" has unmet peer dependency "@babel/core@^7.0.0-0".                                             warning " > @babel/register@7.4.4" has unmet peer dependency "@babel/core@^7.0.0-0".                                                                                                             warning " > @deck.gl/test-utils@7.1.10" has unmet peer dependency "@deck.gl/core@^7.0.0".                                                                                                        warning " > babel-loader@8.0.5" has unmet peer dependency "@babel/core@^7.0.0".                                                                                                                  warning " > babel-loader@8.0.5" has unmet peer dependency "webpack@>=2".                                                                                                                         warning " > eslint-config-uber-jsx@3.3.3" has unmet peer dependency "eslint@>= 3.0.0 < 5".                                                                                                       warning "eslint-config-uber-jsx > eslint-config-uber-es5@2.0.3" has unmet peer dependency "eslint@>= 3.0.0 < 5".                                                                                 warning "eslint-config-uber-jsx > eslint-plugin-react@6.10.3" has unmet peer dependency "eslint@^2.0.0 \|\| ^3.0.0".                                                                               warning " > eslint-plugin-react@7.12.4" has unmet peer dependency "eslint@^3.0.0 \|\| ^4.0.0 \|\| ^5.0.0".                                                                                           warning " > ocular-dev-tools@0.0.27" has incorrect peer dependency "@probe.gl/test-utils@^3.0.2".                                                                                                warning " > @streetscape.gl/layers@1.0.0-beta.16" has unmet peer dependency "@deck.gl/core@^7.1.2".                                                                                              warning " > @streetscape.gl/layers@1.0.0-beta.16" has unmet peer dependency "@deck.gl/layers@^7.1.2".                                                                                            [5/5] Building fresh packages...                                                                                                                                                                 The system cannot find the path specified.                                                                                                                                                       error Command failed with exit code 1.                                                                                                                                                           info Visit https://yarnpkg.com/en/docs/cli/run for documentation about this command.
--

Would you help me to figure it out the issue.

twojtasz, 2019-08-12:
I'd suggest removing the `package-lock.json` file and trying again.  I'm not sure what path is specified, and since it didn't mention the yarn debug log file, i'm not sure where else to look to address this.

twojtasz, 2019-12-10:
Closing this for now. If it is still a problem please re-open.

closed 2019-12-10 by twojtasz