aurora-opensource/streetscape.gl #385: Polylines are distorted in Streetscape compared to other Visualization tool like Kepler or Geojson.io

amolskh, 2019-08-22:
I am trying to visualize below Geojson in Streetscape.gl 
`{
  "type": "FeatureCollection",
  "features": [
    {
      "type": "Feature",
      "properties": {
        "stroke": "#FFFF00",
        "stroke-width": 2,
        "stroke-opacity": 1
      },
      "geometry": {
        "type": "LineString",
        "coordinates": [
                             [
                    -117.1365798525,
                    32.8932626536,
                    0
                  ],
                  [
                    -117.1365903505,
                    32.8932577329,
                    0
                  ],
                  [
                    -117.1366009332,
                    32.89325346,
                    0
                  ],
                  [
                    -117.1366109529,
                    32.8932490706,
                    0
                  ],
                  [
                    -117.1366214774,
                    32.8932444199,
                    0
                  ]
        ]
      }
    }
  ]
}`

I am generating my visualization using Xviz. However polyline looks more zigzag compared to other tools (Kepler, Geojson.io)

Same visualization looks very smooth in **Geojson.io** as below
[image]

Same visualization looks quite smooth in **Kepler.gl**  as below
[image]

However when I visualise same line in **streetscape.gl** it seems to far more distorted as below
[image]

I have observed same behaviour across many polylines.

Is this issue in streetscape.gl implementation? or any config that I need to configure.? Or are other visualisation tools doing any kind of processing that makes line appear more smoother than in case of streetscape.gl ?

amolskh, 2019-08-25:
@twojtasz any thoughts about this issue/ behaviour

Pessimistress, 2019-08-25:
This might be due to xviz parser packing positions into a Float32Arrray. Most of our data is vehicle relative. Lnglat positions would lose precision in 32 bit.

amolskh, 2019-08-26:
Yes I m using JSON format . should I move to GLB ?

amolskh, 2019-08-26:
I moved to GLB dont see any difference .. Could it be that AVS supports higher resolution thank Kepler and other frameworks?

amolskh, 2019-08-28:
I am not sure if this fix is real issue . Because in order to test I pasted same lat long in direct json generated. So Xviz parsing dosent come into play. I simply edited json file created post xviz manually and put the coordinates

amolskh, 2019-08-28:
Ok thanks I will try to test it .. Is this merged into master main release or do I need to take any specific version or using streetscape 1.0 release will work for me ?