aurora-opensource/streetscape.gl #402: how to show road lanes information

EliotCao, 2019-10-25:
I have noted that AVS office website have some more exquisite images as follows,
[usecase-view](https://avs.auto/images/usecase-view.jpg)
[usecase-triage](https://avs.auto/images/usecase-triage.jpg)
[usecase-debug](https://avs.auto/images/usecase-debug.jpg)
Most importantly, it shows more refined environmental information, mainly map information, which can show specific lanes instead of just a road as shown in the website live demo.

So what can I do to achieve the same effect?

Appreciate your work.

twojtasz, 2019-11-11:
@EliotCao It depends on your data.  A lot of autonomy mapping data can be offline & in some custom format, or it could come from the vehicle.  If from the vehicle, you could use XVIZ.
If offline it depends on the data format and how easy it is to convert to something useful.  I do have a GeoJSON example I need to post, that passes a custom deck.gl layer.

Those are the simple cases.  Mapping quickly gets complicated iwth more than just lanes but other layers as well, and as the car moves having those managed in some intelligent way becomes important.

Again, I think the easiest is XVIZ, especially if the data is changing with the vehicle and available.
Next would be some custom layer, such as GeoJSON using deck.gl layers passed in as a custom layer (an example I'll try to get up in the next week or two).

Hope that gives you some ideas.

EliotCao, 2019-11-18:
appreciate it so much.

didibaba, 2020-02-04:
@twojtasz Any progress made so far about the HD map? Is there any examples?

closed 2020-04-27 by EliotCao

heyu3913, 2022-11-04:
> @twojtasz Any progress made so far about the HD map? Is there any examples?

+1