aurora-opensource/streetscape.gl #410: Primitive type icon is not shown if the metadata is generated by XVIZMetadataBuilder

lgmcode, 2019-11-15:
Hi, I try to run website-demo, and it works well if I just use the KITTI dataset. But when I use XVIZMetadataBuilder to generate a metadata message myself, the primitive type icon will not display. I console.log the meta message and compare it with the meta part of `1-frame.glb`, found that the `scalar_type` is in upper case in the generated one and lower case in the `.glb` file. Also found in the `custom-style.js` the icon object `PRIMITIVE_TYPE_TO_ICON` has all lower case keys. I think this may be the reason why type icons are missing. How to generate a meta message with lower case `scalar_type`?

twojtasz, 2019-12-05:
@lgmcode This seems to be a problem in the XVIZ Builders.  Let me take a look.

twojtasz, 2019-12-06:
@lgmcode Actually, we did change enums to UPPERCASE.  So the builder is correct, the website-demo is simply not up to date.  I will change the demo code to apply `.toUpperCase()` to the type.

twojtasz, 2019-12-10:
This fix was addressed in https://github.com/uber/streetscape.gl/pull/414

closed 2019-12-10 by twojtasz