aurora-opensource/streetscape.gl #411: utils/view-port.js overwrites transitionDuration when not in firstPerson viewMode

jorisw, 2019-11-25:
Hi,

Today I was trying to utilize deck.gl's `transitionDuration` option for view states, to be able to switch view mode with a nice camera transition. However my setting wouldn't stick, and tracing the option through the LogViewer source, I found that the setting is being overwritten here:

https://github.com/uber/streetscape.gl/blob/6a3549eba1ff122fe2135c87d576ac4dbe6ee255/modules/core/src/utils/viewport.js#L114

It seems like it only allows for use of the option when `viewMode.firstPerson` is true. Is there some special reason that I shouldn’t use transitionDuration except when in `firstPerson` view mode?

twojtasz, 2019-12-05:
@Pessimistress any idea why we would force transitionDuration here?  Does it cause problems with camera in any way?

jorisw, 2019-12-13:
Upon editing `viewport.js` to stop overwriting the `transitionDuration`, it appears that Deck.gl makes React crash writing the transition updates to viewState. I've been trying to resolve this but I can't seem to, without disabling the view state updates that are part of the transition entirely. I can't figure out if this is a Streetscape problem or a Deck problem.

```
Error: Maximum update depth exceeded. This can happen when a component repeatedly calls setState inside componentWillUpdate or componentDidUpdate. React limits the number of nested updates to prevent infinite loops.
    at checkForNestedUpdates (react-dom.development.js:25671)
    at scheduleUpdateOnFiber (react-dom.development.js:23674)
    at Object.enqueueSetState (react-dom.development.js:13994)
    at LogViewer../node_modules/react/cjs/react.development.js.Component.setState (react.development.js:325)
    at Object.onViewStateChange (index.js:89)
    at Object.onViewStateChange (core-3d-viewer.js:182)
    at Deck._onViewStateChange (deck.js:738)
    at ViewManager._onViewStateChange (view-manager.js:247)
    at TransitionManager._onTransitionUpdate (transition-manager.js:163)
    at Transition.update (transition.js:81) {componentStack: "↵    in DeckGL (created by Core3DViewer)↵    in Co…ed by App)↵    in div (created by App)↵    in App"}
```