aurora-opensource/streetscape.gl #415: Cannot seen to pre-set viewOffset

jorisw, 2019-12-10:
Whenever I pass a pre-set `viewOffset` to `<LogViewer />`, like so:

```
<LogViewer
      log={log}
      mapboxApiAccessToken={MAPBOX_TOKEN}
      mapStyle={MAP_STYLE}
      car={CAR}
      xvizStyles={XVIZ_STYLE}
      showTooltip={settings.showTooltip}
      viewMode={VIEW_MODE[settings.viewMode]}
      viewOffset={{x: 1, y: 400, bearing: 1}}
/>
```

... the `y` setting is ignored until I drag the viewport, while `bearing` is picked up and reflected before I do anything.

All I can find to cause this, is the blind overwriting of `offset` in `core/src/utils/viewport.js` :

This happens to me when running either apps in `test/apps/viewer/` and `examples/getting-started/`.

https://github.com/uber/streetscape.gl/blob/6a3549eba1ff122fe2135c87d576ac4dbe6ee255/modules/core/src/utils/viewport.js#L117-L124

Streetscape.gl 1.0.0 - 1.0.2.

jorisw, 2019-12-10:
I've debugged the block of code I'm citing above and it's not the cause, in such that the offset.x and offset.y are retained past that block. A console.log reveals the values that I'm passing with `viewOffset`, yet the viewport only reflects offset on drag.

jorisw, 2019-12-10:
Fixed if I pass `width` and `height` properties with our `initialViewState` to match the viewport, e.g. `window.innerHeight`, `window.innerWidth`.

pai-qi, 2021-08-17:
> Fixed if I pass `width` and `height` properties with our `initialViewState` to match the viewport, e.g. `window.innerHeight`, `window.innerWidth`.

Thanks