aurora-opensource/streetscape.gl #431: 'Driver' view mode causes a crash, streetscape.gl versions 1.0.3 and 1.0.4

nwakkaf, 2020-02-24:
Hello, 

When trying to run streetscape frontend and switch to driver view, a crash happens once we start to move the camera with the message "Uncaught Error: failed to invert matrix".
This is maybe related to the upgrade of deck.gl (or other dependencies) in streetscape version 1.0.3 that have introduced this bug.

Can you please help with the fix, thanks!

didibaba, 2020-03-04:
#427

nalini0505, 2020-03-26:
Hi, If I change streetscape version to 1.0.2 will it work?

didibaba, 2020-04-04:
@nalini0505 
Yes, it is exactly what I have done.

xstmjh, 2020-08-21:
It's still an issue in the latest version

xhlulu, 2020-09-09:
> It seems that onViewStateChange prop of DeckGL component does not return 'zoom' value on DRIVER mode (i.e. firstPerson). Zoom value is, however, expected by viewState prop.
> A temporary solution would be to define own Driver mode with some zoom level (inside firstPerson).
> A fix might be to update _onViewStateChange of LogViewer to accept DEFAULT_VIEW_STATE beforehand to setting the new incoming viewState.

Could you share code or a gist of your updated `app.js` showing how that would work? I'm interested by this solution and I'm not sure whether this problem will be solved by the authors.

nwakkaf, 2020-09-13:
> Could you share code or a gist of your updated `app.js` showing how that would work? I'm interested by this solution and I'm not sure whether this problem will be solved by the authors.

You can maybe have something like this (in map-view.js render()):
```
if (VIEW_MODE[settings.viewMode] && VIEW_MODE[settings.viewMode].firstPerson) {
    VIEW_MODE[settings.viewMode].firstPerson["zoom"] = 20; // inject some value
}
```
Or copy VIEW_MODE and alter DRIVER.firstPerson by adding some zoom level.