aurora-opensource/streetscape.gl #435: set uniform color to all points

srikanth-kuriyala, 2020-04-06:
I'm trying to set the uniform color to lidar points displayed.

Created an array and pushed the same color [255, 255, 0] in it using loop.
```
var arr = [];
for(let i=0; i<=data[0].points.length; i++){
            arr.push(255, 255, 0);
}
colors = new Uint8ClampedArray(arr);
return new PointCloudLayer(
          forwardProps,
          layerProps,
          this.getSubLayerProps({
            id: 'pointcloud',
            data: {
              length: data[0].points.length / 3,
              attributes: {
                getPosition: data[0].points,
                getColor: colors
              }
            },
            vehicleRelativeTransform: this.props.vehicleRelativeTransform,
            getPosition: p => p
          })
        );
```
With approach it's mixing up with the red color also in the points.

Can someone suggest me where am doing wrong.

Thank you.

ibgreen, 2020-04-06:
You are probably leaving out the alpha value - colors are typically `RGBA`. Try:

```
for(let i=0; i<=data[0].points.length; i++){
            arr.push(255, 255, 0, 255);
}
```

srikanth-kuriyala, 2020-04-07:
> You are probably leaving out the alpha value - colors are typically `RGBA`. Try:
> 
> ```
> for(let i=0; i<=data[0].points.length; i++){
>             arr.push(255, 255, 0, 255);
> }
> ```

Thanks a lot. It's working.

closed 2020-04-07 by srikanth-kuriyala