aurora-opensource/streetscape.gl #443: Issue with playing nuscenes data

KSaikishore, 2020-05-13:
I have converted the nuscenes data(mini,Depreciated datasets) to xviz.But when i visualised that on streetscape.gl camera and lidar data were not getting visualised properly(data is dissapearing) most of the time.How do i rectify that

closed 2021-02-15 by KSaikishore

malao98, 2021-09-06:
Hi, 

I met the same problem as yours when im trying to visualize nuscenes dataset using streetscape gl. Both my lidar data and camera image are not working too. Did you figure out how to solve it?

KSaikishore, 2021-09-06:
> Hi,
> 
> I met the same problem as yours when im trying to visualize nuscenes dataset using streetscape gl. Both my lidar data and camera image are not working too. Did you figure out how to solve it?

nope