aurora-opensource/streetscape.gl #446: How to realize the bubble icon with bike/car ?

baker-Xie, 2020-06-02:
Dear all, this is a question not a issue. I'm new to WebGL, and I want to know how this repo realize the following effect.
[image]
I find that in streetscape repo, there is a streetscape.svg file that contain the bike svg path. But I cannot find the bubble, and don't know to realize it.
So I have two questions:
1. Does bubble and bike have seperate svg path/file?
2. How to show them in the 3D canvas? Using <div>? or Something like Sprite in three.js?

soyons, 2023-05-25:
Has anybody can show me how to realize this bubble effect? I'm also instested in it.

soyons, 2023-05-26:
you need to wirte something in function ````renderObjectLabel``` and add another stream with the same id of obstacles.