aurora-opensource/streetscape.gl #448: Zoom in/out Feature

ravi1122, 2020-06-20:
Hi All,

I am trying to implement Zoom in/out feature which mouse scrolling wheel is taking as the center of the windows screen rather than its default pointer as the center, I want windows screen should be center and whenever we will zoom in/out it will take window screen should be center rather than mouse point. Is there any way we can do it? Can anyone help me regarding the same?

jlisee, 2020-06-30:
I don’t know this area of the code but at one time the default was the way you describe and we changed it.  So searching the history can help, it’s a standard deck.gl setting.

ibgreen, 2020-06-30:
Not sure if there is a [controller setting](https://deck.gl/docs/api-reference/core/controller), if not you may be able to create a custom deck.gl controller. 

You can always open a question in the deck.gl repo.