aurora-opensource/streetscape.gl #450: Polyline does not appear when using GEOGRAPHIC coordinate

kgreenek, 2020-07-02:
Might be related to #307 

I am using the python builder library for the xviz server. I have verified that the json state_update sent to streetscape looks as expected.

This is how I set the metadata for the stream:

```python
builder.stream("/foo/bar")\
    .category(xviz_avs.CATEGORY.PRIMITIVE)\
    .type(xviz_avs.PRIMITIVE_TYPES.POLYLINE)\
    .coordinate(xviz_avs.COORDINATE_TYPES.GEOGRAPHIC)\
    .stream_style({
        'stroked': True,
        'stroke_width': 0.2,
        'stroke_color': [0, 255, 0, 128]
    })
```

This is how I create the state_update:

```python
builder.primitive("/foo/bar").polyline([lon1, lat1, 0.0, lon2, lat2, 0.0])
```

If I set the coordinate to VEHICLE_RELATIVE, I can render lines just fine.

kgreenek, 2020-07-02:
If I change the polyline to be points instead and use the same GPS coordinates, the points render but the polyline does not.

pai-qi, 2021-07-19:
I have the same problem, when the distance between the points is too short, it will appear