aurora-opensource/streetscape.gl #465: how to integrate nebula gl with streetscape.gl ?

aubeena007, 2020-09-10:
i want introduce drawing in streetscape.gl so am planing to add nebula gl how to this ?

ibgreen, 2020-09-10:
Two key observations would be
- streetscape.gl is based on deck.gl and allows the application to add its own custom deck.gl layers.
- nebula.gl is a layer pack for deck.gl with a number of editable deck.gl layers.
i.e. you should be able to pass nebula.gl layers to streetscape.gl.

aubeena007, 2020-09-14:
@ibgreen  i added nebula gl layer pack but it is not sync with pointcoud layer what will be the issue 
attaching a screen-cast  
[image]

ibgreen, 2020-09-14:
Hmm, hard to say. 

There may be issues with the coordinate systems, and nebula.gl support for` COORDINATE_SYSTEM.METER_OFFSETS` 

Do you add any other custom deck.gl layers (non-editable)? Do those work correctly?

closed 2020-09-16 by aubeena007