aurora-opensource/streetscape.gl #473: About Streetscape Performance Metrics

BupyeongHealer, 2020-12-04:
Hello! 
Our team uses xviz and streetscape.gl to create a visualization system.
Performance metrics are now used to determine latency rates.
However, I do not understand the meaning of xviz frames loaded, xviz frames lendered. What is that?
Also, the Our ros topics 10hz(gps) but xviz frames rendering degree is 22.
Why are these two figures different?

[image]

twojtasz, 2020-12-07:
Hopefully the documentation is clear:
https://avs.auto/#/streetscape.gl/api-reference/react-components/log-viewer?section=debug-function-optional-

But to add additional context, frames are basically XVIZ Messages, specifically **StateUpdate**.  So it should be the number of messages we received.

The XVIZFrames Rendered matches the Redraw, which is **less** than the FPS.  What I believe this means is that the FPS is the  raw redraw performance, so how many times the webgl context was drawn regardless of the actual data changes.  So i believe FPS should always be >= Redraw.

XVIZ Frames Rendered should be the **actual** different set of XVIZ data that was drawn.  What this means is that if the data is the same across 2 renderings, XVIZ Frame Renders will not increment, but your FPS metric would increase. 

It looks like your messages received are 87, but we have only rendered 22.  That could be due to a various reasons.  The metrics are very basic events. We rendered via GPU, we updated the scene due to new data, we have seen  a message. What that doesn't tell you is what messages have been rendered because it depends on a few other factors.

BupyeongHealer, 2020-12-07:
@twojtasz Thank you for your kind explanation. I understand clearly!

closed 2020-12-07 by BupyeongHealer