aurora-opensource/streetscape.gl #477: Map support with higher pitch angles

nwakkaf, 2021-01-04:
Hello,

Will you consider upgrading to Mapbox version 2.0.0 (https://github.com/mapbox/mapbox-gl-js/releases/tag/v2.0.0) where the camera pitch angle is unlocked?

Thanks

twojtasz, 2021-01-04:
The new version has a new license right.  I need to see what our dependency lib does, react-map-gl.

https://github.com/visgl/react-map-gl/blob/master/docs/get-started/mapbox-tokens.md

I'm not sure just yet how to best approach this.  Ideally it's a user choice with sensible defaults.  If I can find a way to have the mapbox v1 as the default, with the ability to pass in or configure the v2 (all via react-map-gl so really v5 & v6 of that project), then that would make the choice moot.

nwakkaf, 2021-01-05:
We actually use streetscape as a dependency for now (no access to its internal code). So yes if it would be possible to have the Map version configurable at the application top-level (user's side), then this would meet our need (we can decide then whether to go with the commercial v2 Mapbox license or just keep using the older version)

I see this discussion, however, (https://github.com/visgl/deck.gl/discussions/5237) and wonder whether the Mapbox support as a base map will be dropped in the future!

Thanks for your response and support.

ibgreen, 2021-01-06:
> I see this discussion, however, (visgl/deck.gl#5237) and wonder whether the Mapbox support as a base map will be dropped in the future!

No plans to drop mapbox support. The goal of deck.gl to support the best commercial base map integrations. Mapbox is likely to remain a great base map option for users who need very high quality base maps and are willing to pay for that. 

That said, deck.gl would like to offer a free, lower quality base map option in parallel. I don't want to go ahead of things for deck.gl itself, but for now, if you follow the updates in react-map-gl docs http://visgl.github.io/react-map-gl/docs you can see work that is being done there to support alternative basemaps.

nwakkaf, 2021-01-07:
Hmm, sounds good, thanks for the illustration.