aurora-opensource/streetscape.gl #478: License for xviz-data repo

nwakkaf, 2021-01-15:
Hello,

I checked out this repo (https://github.com/uber/xviz-data.git). I want to mirror it and be able to run streetscape's website-demo.
However, there is no license file bound with that repo, is it left out on purpose? can we alter its content after mirroring it?

Thanks

twojtasz, 2021-01-22:
@nwakkaf No need to really mirror, that data was generated using the public KITTI data set and is the same that is used in the getting-started example.

So you could just generate your self, the KITTI data i believe is a open license (you must verify) so you may need to to cite that as appropriate.  In ours demo app there is a citation visible in the application info tab.

And you need to verify for your usage, but I recall we could "transform" the data, but there are classes of "edits" that might be outside the license.

nwakkaf, 2021-01-27:
Thanks for your answer and the info and sorry for my late response.

Actually, my idea was to mirror that repo and remove the 'nutonomy' dataset and keep testing with KITTI only!
However, being able to use XVIZFileLoader (and load data on-demand) would solve my issue.

Kind regards

closed 2022-01-02 by twojtasz