aurora-opensource/streetscape.gl #480: TypeError: Super expression must either be null or a function

nasrat-v, 2021-01-29:
I recently have this error without changing anything in my code.

_**Error**_

`> Build error occurred
TypeError: Super expression must either be null or a function
    at _inherits (.../node_modules/@babel/runtime/helpers/inherits.js:5:11)
    at .../node_modules/@streetscape.gl/core/dist/es5/components/log-viewer/perspective-popup.js:102:29
    at Object.<anonymous> (.../node_modules/@streetscape.gl/core/dist/es5/components/log-viewer/perspective-popup.js:204:2)
    at Module._compile (internal/modules/cjs/loader.js:1256:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1277:10)
    at Module.load (internal/modules/cjs/loader.js:1105:32)
    at Function.Module._load (internal/modules/cjs/loader.js:967:14)
    at Module.require (internal/modules/cjs/loader.js:1145:19)
    at require (internal/modules/cjs/helpers.js:75:18)
    at Object.<anonymous> (.../node_modules/@streetscape.gl/core/dist/es5/components/log-viewer/object-labels-overlay.js:30:48) {
  type: 'TypeError'
}
`

_**Environment**_

- streetscape.gl v1.0.6
- nextJS` v9.5.3
- node v14.7.0
- npm v6.14.7

**_Steps to Reproduce_**

- `npm run build`
The error happened only during build in production.

It seems related to a package downloaded by streetscape, because it has worked without any changes locally.

I tried differents webpack configs, like `keep_fnames: true` on the minimizer plugin `Terser` or `UglifyJS` 
like this issue [https://github.com/facebook/create-react-app/issues/7236](url) but none worked.

cmpute, 2021-01-31:
I can confirm this issue when I completed a fresh install of streetscape and run the example. This error is shown in the debug console.

*Env*:
- Ubuntu 18.04
- streetscape.gl master
- node v10.23.2
- npm v6.14.10

nasrat-v, 2021-02-02:
The error seems only related to Linux.
`npm run build` run successfully on windows 10 with same version of node and npm.

gitpayal, 2021-02-02:
I am getting this error   with the windows machine
**Environment :**
npm 6.14.10
node v14.15.4
streetscape.gl latest
windows 10 enterprise

**Error** 

node_modules/@babel/runtime/helpers/esm/inherits.js:4
  1 | import setPrototypeOf from "@babel/runtime/helpers/esm/setPrototypeOf";
  2 | export default function _inherits(subClass, superClass) {
  3 |   if (typeof superClass !== "function" && superClass !== null) {
> 4 |     throw new TypeError("Super expression must either be null or a function");
  5 |   }
  6 | 
  7 |   subClass.prototype = Object.create(superClass && superClass.prototype, {
View compiled

Module../node_modules/@streetscape.gl/core/dist/esm/components/log-viewer/perspective-popup.js
C:/Users/^^^^^/src/components/log-viewer/perspective-popup.js:64
  61 | }));
  62 | 
  63 | /* Like Popup but deal with z */
> 64 | class PerspectivePopup extends Popup {
  65 |   _renderTip(positionType) {
  66 |     const anchorPosition = ANCHOR_POSITION[positionType];
  67 |     const {theme, style} = this.props;

**Steps to Reproduce**

npm run build
npm start

**package.json**

  ```
"dependencies": {
    "@loaders.gl/obj": "^2.2.8",
    "antd": "^4.8.4",
    "file-loader": "^6.0.0",
    "history": "^4.10.1",
    "http-proxy-middleware": "^1.0.6",
    "jwt-decode": "^2.2.0",
    "patch-package": "^6.2.2",
    "react": "^17.0.1",
    "react-dom": "^17.0.1",
    "react-router-dom": "^5.2.0",
    "react-scripts": "4.0.0",
    "react-toast-notifications": "^2.4.0",
    "rxjs": "^6.6.3",
    "streetscape.gl": "^1.0.0",
    "web-vitals": "^0.2.4"
  },

  "devDependencies": {
    "@testing-library/jest-dom": "^5.7.0",
    "@testing-library/react": "^10.0.4",
    "@testing-library/user-event": "^10.1.2",
    "@types/classnames": "^2.2.10",
    "@types/color": "^3.0.1",
    "@types/flat": "^5.0.1",
    "@types/jest": "^26.0.4",
    "@types/lodash": "^4.14.158",
    "@types/node": "^14.0.20",
    "@types/react": "^16.9.41",
    "@types/react-dom": "^16.9.8",
    "@types/react-router-dom": "^5.1.5",
    "@typescript-eslint/eslint-plugin": "^3.6.1",
    "@typescript-eslint/parser": "^3.6.1",
    "cross-env": "^7.0.2",
    "css-loader": "^3.4.2",
    "cypress": "^6.3.0",
    "eslint": "^7.11.0",
    "eslint-config-prettier": "^6.11.0",
    "eslint-plugin-prettier": "^3.1.4",
    "eslint-plugin-react": "^7.19.0",
    "husky": "^4.2.5",
    "jest-junit": "^10.0.0",
    "node-sass": "^4.9.0",
    "prettier": "^2.0.5",
    "sass-loader": "^7.0.1",
    "source-map-loader": "^0.2.3",
    "style-loader": "^0.21.0",
    "typescript": "^3.9.6"
  }
```

Any suggestions about it

cmpute, 2021-02-03:
> The error seems only related to Linux.
> `npm run build` run successfully on windows 10 with same version of node and npm.

Could you please share the npm package versions on windows, so that we can compare and see whether it's due to changes of dependent packages?

nasrat-v, 2021-02-03:
I confirm that I can build and start on Windows 10 Education 20H2.

**_Environment_**
- node 14.15.4
- npm 6.14.10
- streetscape 1.0.6
- reactjs 16.13.1
- nextjs 9.5.3

_**package.json**_
````
"dependencies": {
    "@date-io/date-fns": "^1.3.13",
    "@loaders.gl/core": "^2.3.6",
    "@loaders.gl/obj": "^2.3.6",
    "@material-ui/core": "4.11.0",
    "@material-ui/data-grid": "4.0.0-alpha.6",
    "@material-ui/pickers": "^3.2.10",
    "@react-google-maps/api": "^1.13.0",
    "@types/node": "^14.14.16",
    "axios": "0.20.0",
    "bootstrap": "4.5.2",
    "bowser": "^2.11.0",
    "bufferutil": "^4.0.2",
    "date-fns": "^2.16.1",
    "eslint": "^7.12.1",
    "eslint-plugin-react": "^7.21.5",
    "formik": "2.1.5",
    "fs": "0.0.1-security",
    "js-cookie": "2.2.1",
    "jwt-decode": "3.0.0-beta.2",
    "moment": "^2.28.0",
    "next": "9.5.3",
    "next-cookies": "2.0.3",
    "next-fonts": "1.4.0",
    "next-redux-wrapper": "^6.0.2",
    "pubnub": "^4.29.10",
    "react": "16.13.1",
    "react-bootstrap": "1.3.0",
    "react-bootstrap-modal": "^4.2.0",
    "react-dom": "16.13.1",
    "react-dropzone": "11.2.0",
    "react-geocode": "^0.2.2",
    "react-inlinesvg": "2.1.1",
    "react-perfect-scrollbar": "1.5.8",
    "react-redux": "^6.0.0",
    "react-spinners": "0.9.0",
    "react-truncate": "2.4.0",
    "reactstrap": "^8.8.1",
    "recharts": "1.8.5",
    "redux": "4.0.5",
    "redux-devtools-extension": "^2.13.8",
    "redux-thunk": "^2.3.0",
    "regl-worldview": "^0.14.1",
    "resize-observer-polyfill": "^1.5.1",
    "sass": "^1.30.0",
    "scaledrone-node": "^1.5.0",
    "scaledrone-react-native": "^1.5.0",
    "sockjs-client": "^1.5.0",
    "streetscape.gl": "^1.0.6",
    "swr": "0.3.3",
    "utf-8-validate": "^5.0.3",
    "ws": "^7.4.1",
    "yup": "0.29.3"
  },
  "devDependencies": {
    "@testing-library/dom": "7.24.2",
    "@testing-library/jest-dom": "5.11.4",
    "@testing-library/react": "11.0.4",
    "@testing-library/user-event": "12.1.5",
    "babel-jest": "26.3.0",
    "canvas": "2.6.1",
    "eslint": "^7.12.1",
    "eslint-plugin-react": "^7.21.5",
    "http-proxy-middleware": "^1.0.6",
    "identity-obj-proxy": "3.0.0",
    "jest": "26.4.2",
    "jest-dom": "4.0.0",
    "loaders.gl": "^0.3.5",
    "redux-mock-store": "^1.5.4"
  }
````

Pessimistress, 2021-02-03:
This is caused by react-map-gl switching to functional components. You can work around this with the following in package.json:

```
"resolutions": {
  "react-map-gl": "5.2.12"
}
```

nasrat-v, 2021-02-03:
> This is caused by react-map-gl switching to functional components. You can work around this with the following in package.json:
> 
> ```
> "resolutions": {
>   "react-map-gl": "5.2.12"
> }
> ```

Thanks you ! 
It fixed the build
Don't forget that `"resolutions"` only work with `yarn`

twojtasz, 2021-02-05:
I need to publish a new release, then it should work