aurora-opensource/streetscape.gl #484: import 'fs' library does not work

EdoardoSerri, 2021-03-08:
Hi everybody,

I have an issue that I do not know if it is a bug or else. 
I'm trying to import the library 'fs' to iterate through the file system and dynamically choose the length of the LOGS array, in constant.js, to be able to visualize more than one XVIZ file, not knowing the name of the number of the XVIZ files I have to display.

However, I have installed 'fs' using : 
 `npm install fs` 
imported in constant.js using:
`var fs=require('fs')`  

it gives me out this error
 `Module not found: Error: Can't resolve 'fs' in 'C:\Users\tdro5q\streetscape.gl\examples\website-demo\src\constant.js`

I have:
- Win 10
- Node v14.15.4
- streetscape.gl master

Thank you for the help

ibgreen, 2021-03-08:
Probably stating he obvious, `fs` is a node.js library, that is not available in the browser. 

Are you using it on the server side?

EdoardoSerri, 2021-03-08:
@ibgreen yes. I'm not so confident with javascript. So I may be mistaking something in the workflow.
That's it

- import the fs library in the javascript file
- npm install fs
- yarn 
- yarn start-local

Am I doing something wrong?

closed 2021-03-11 by EdoardoSerri