aurora-opensource/streetscape.gl #493: $ yarn -- error Command failed. Exit code: 128

yanganlan0310, 2021-04-18:
error Command failed.
Exit code: 128
Command: git
Arguments: ls-remote --tags --heads https://github.com/jlisee/protobuf.js.git
Directory: /home/yang/software/streetscape.gl/examples/get-started
Output:
fatal: unable to access 'https://github.com/jlisee/protobuf.js.git/': Failed to connect to github.com port 443: Connection refused
info Visit https://yarnpkg.com/en/docs/cli/install for documentation about this command.
Error: Command failed.
Exit code: 128
Command: git
Arguments: ls-remote --tags --heads https://github.com/jlisee/protobuf.js.git
Directory: /home/yang/software/streetscape.gl/examples/get-started
Output:
fatal: unable to access 'https://github.com/jlisee/protobuf.js.git/': Failed to connect to github.com port 443: Connection refused
    at ProcessTermError.ExtendableBuiltin (/mnt/c/Users/Lenovo/AppData/Roaming/npm/node_modules/yarn/lib/cli.js:721:66)
    at ProcessTermError.MessageError (/mnt/c/Users/Lenovo/AppData/Roaming/npm/node_modules/yarn/lib/cli.js:750:123)
    at new ProcessTermError (/mnt/c/Users/Lenovo/AppData/Roaming/npm/node_modules/yarn/lib/cli.js:790:113)
    at ChildProcess.<anonymous> (/mnt/c/Users/Lenovo/AppData/Roaming/npm/node_modules/yarn/lib/cli.js:25884:17)
    at ChildProcess.emit (events.js:314:20)
    at maybeClose (internal/child_process.js:1022:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:287:5)

environment: WSL(win10,ubutun18.04);
nodejs --v12.22.1
yarn -- v1.22.10
npm --v6.14.12
thank you !

hujianglong, 2021-05-14:
same issue, Is there a solution to this problem

hujianglong, 2021-05-14:
I try use this command, and it is ok for this problem:
git config --global url.git://github.com/.insteadOf https://github.com/

didibaba, 2022-12-28:
@hujianglong  Still not working. Any other options to solve this problem?

ThierryBayala, 2023-02-11:
@didibaba did you get any solution to fix it? I am having the same problem

okay00, 2023-02-19:
@ThierryBayala you get any solution

fang-hd, 2023-05-25:
how to solve this problem

okay00, 2023-05-25:
Missing or incorrect installation: Ensure that you have Yarn installed on your system and that it is up to date. You can check your Yarn version by running yarn --version in your terminal. If Yarn is not installed, you can follow the official Yarn installation guide to install it correctly.

Incorrect command or arguments: Verify that you are running the correct Yarn command and providing the appropriate arguments. Check for any typos or incorrect syntax in the command you're trying to execute.

Permission issues: Ensure that you have the necessary permissions to run the yarn command. If you are facing permission issues, you may need to run the command with administrative privileges or adjust the permissions on the relevant directories.

Network connectivity: If the command requires an internet connection, ensure that you have a stable internet connection and that there are no network issues or firewalls blocking the Yarn command's access to the internet.

okay00, 2023-05-25:
@fang-hd