aurora-opensource/streetscape.gl #502: Importing HD map in XODR format or a map from openstreet maps

AyaYaf, 2021-05-30:
I have a question regarding streetscape.gl and would appreciate your help.

Can we import HD map in XODR format or a map from openstreet maps in the Log Viewer?

Thank you

yujingz, 2021-06-02:
I'd say the easiest way is to convert them into geojson

pai-qi, 2021-06-15:
> I'd say the easiest way is to convert them into geojson

+1

JianXinyu, 2021-12-13:
> I'd say the easiest way is to convert them into geojson

Could you please give some advice on how to convert the XODR format into GeoJson?  Thanks!

yujingz, 2021-12-14:
> > I'd say the easiest way is to convert them into geojson
> 
> Could you please give some advice on how to convert the XODR format into GeoJson? Thanks!

try https://sumo.dlr.de/docs/netconvert.html

JianXinyu, 2021-12-15:
> > > I'd say the easiest way is to convert them into geojson
> > 
> > 
> > Could you please give some advice on how to convert the XODR format into GeoJson? Thanks!
> 
> try https://sumo.dlr.de/docs/netconvert.html

Thanks! I'll this one. Also, I found a repo that can directly transform .xodr into GeoJson. See [this repo](https://github.com/JianXinyu/OpendriveParser) that I forked from the original repo.

xiaokn, 2023-03-08:
> I'd say the easiest way is to convert them into geojson

> I'd say the easiest way is to convert them into geojson

I found that geojson and mapbox do not match visually in height, have you encountered a similar problem?