aurora-opensource/streetscape.gl #505: glb load URL with query parameters

vince52, 2021-07-09:
Hello,

I tried to load a .glb file from network but I had an issue.
The URL have some query parameters so it couldn't be parse.

An example of an URL with query parameter that cannot be parse:
https://github.com/uber/xviz-data/blob/master/kitti/2011_09_26_drive_0005_sync/1-frame.glb?parameter=something

I saw that the way used to parse the file is here:
https://github.com/uber/streetscape.gl/blob/759fb75db2a60a3ed0e8820d16fc1a18c94e59ee/modules/core/src/loaders/xviz-file-loader.js#L105-L106

If I modify the `fileFormat` by adding something to remove query parameter, it seems to work.

Thank you for your help.

yujingz, 2021-08-05:
First of all, this is not related to the repo. Consider closing this issue and ask on stackoverflow. 

Back to the question, you are using github as a file server and in a wrong way. Try something like https://github.com/uber/xviz-data/blob/master/kitti/2011_09_26_drive_0005_sync/1-frame.glb?raw=true to get the raw file instead of the github page.

vince52, 2021-08-05:
Sorry if there is a misunderstanding.
I'm using AWS S3 to host XVIZ files.
I was only using GitHub XVIZ demo repository to show the issue behind it.

The issue is that if you put any query behind the file format, and if you use file way of reading files, the core module will fail reading the glb file.

Sorry if it's not a correct issue. 
I discovered another way to do it using stream, so I will continue on this side.

Thank you.

closed 2021-08-05 by vince52