aurora-opensource/streetscape.gl #508: Is it possible to draw a polygon using center`s gps and  long\width\hight ?

pai-qi, 2021-08-04:
I konw it can be done using 4 gps  , and I generate the 4 gps with center`s gps + long\width\height, but it takes too much time, is there another way I don`t know ? Thanks

yujingz, 2021-08-05:
I think you are confusing rectangle/square with polygon, if you think carefully about the definition of polygon, you won't ask this question. My suggestion would be to make a custom layer based off of the polygon layer

Also I don't understand why calculating vertices from center + length, width, and height would be time-consuming, do you have so many bounding boxes? If you care so much about performance, prepare the data in the backend before it hits the front end

pai-qi, 2021-08-06:
> I think you are confusing rectangle/square with polygon, if you think carefully about the definition of polygon, you won't ask this question. My suggestion would be to make a custom layer based off [...]

Thanks for your reply. 
What I need is to draw the cars on the road, so my first choice is to use XVIZMetadataBuilder. What I get from datasource is a car`s center gps and its length\width\height,  this api seems  not to support it without coordinate transformation（which tackes much time either in frontend or in backend)?

twojtasz, 2022-01-02:
The `/vehicle_pose` represents a point.  The default mesh we create this point at the center of the rear axle, however, you can provide your own `car` prop (see https://avs.auto/#/streetscape.gl/api-reference/react-components/log-viewer?section=car-object-optional-)

And if you prep mesh you can create it as necessary to match your use-case.

The only case that would be awkward is if you have a number of different sizes you need.  However, Im sure we can advise a way around that if it was the case.