aurora-opensource/streetscape.gl #511: am getting error when running example site

aubeena007, 2021-09-02:
am getting error when running example site.

WARNING in ./node_modules/@xviz/io/dist/esm/gltf/glb-builder.js 98:25-36
"export 'padTo4Bytes' was not found in '@loaders.gl/loader-utils'

twojtasz, 2022-01-02:
I"m working to update this issue. it's from an out dated dependency library that seems to have broken semver.

twojtasz, 2022-01-02:
I'm not able to reproduce as the yarn.lock file has @loaders.gl/loader-utils  at version 2.3.13, which has the padTo4Bytes.  However, it was pointed out this dependency is missing in the package.json for @xviz/io.  So i'm adding that in hope it addresses the issue.  I'm guessing since it was not listed it was getting v3 and thus causing the problem.

updating to loaders.gl v3 can come next.

twojtasz, 2022-01-03:
finally able to repro, comes from newer deck.gl which includes loaders.gl 3.  Just running the  the get-started example with `yarn start-streaming-local` was able to show this error in the build output.

FFEverglow, 2022-04-08:
doesn't work