aurora-opensource/streetscape.gl #514: Obtain the coordinates of four points in the visual range

juzi5230, 2021-11-01:
Can I obtain the coordinates of four points in the visual range

yujingz, 2021-11-04:
Can you elaborate on what the visual range means?

juzi5230, 2021-11-09:
> 你能详细说明视觉范围是什么意思吗？

> Can you elaborate on what the visual range means?

> Can you elaborate on what the visual range means?

The range we can see of the screen,
 i want to obtain the coordinates of the four vertices in the screen,
 or the minimum coordinate point and the maximum coordinate point

juzi5230, 2021-11-09:
> Can you elaborate on what the visual range means?

The range we can see of the screen,
 i want to obtain the coordinates of the four vertices in the screen,
 or the minimum coordinate point and the maximum coordinate point

yujingz, 2021-11-09:
Coordinates doesnt really make sense in 3d views

For 2D views, check out https://avs.auto/#/streetscape.gl/api-reference/react-components/log-viewer?section=viewstate-object-optional-

juzi5230, 2021-12-01:
> Coordinates doesn't really make sense in 3d views
> 
> For 2D views, check out https://avs.auto/#/streetscape.gl/api-reference/react-components/log-viewer?section=onclick-function-optional-

I get it ,thanks for you reply！

twojtasz, 2022-01-02:
@juzi5230 I think what you want are the bounds of the viewport.  deck.gl supports this, see https://deck.gl/docs/api-reference/core/viewport

I don't know if we expose this at the. moment.