aurora-opensource/streetscape.gl #522: How to add two ego cars?

JianXinyu, 2021-12-30:
Is it possible to show another car using the same car model as the ego car? I'd like to show a shadow-ego car as the historical trajectory.

twojtasz, 2022-01-02:
@JianXinyu One way we approached this was simply to define a pose, such as `/ghost_pose`, but then we added them directly as a `customLayer` to the LogViewer props.

JianXinyu, 2022-01-06:
@twojtasz Thanks for your answer! I'll test both methods in my project.

SlahserZ, 2022-01-10:
```
const customLayers = [ new SimpleMeshLayer({
        id: 'shadow-ego',
        opacity: 0.5,
        coordinateSystem: COORDINATE_SYSTEM.METER_OFFSETS,
        coordinateOrigin: matrix?.origin || CAR.origin,
        // Adjust for car center position relative to GPS/IMU
        getTransformMatrix: d =>
          matrix?.vehicleRelativeTransform
            ?.clone()
            ?.translate(origin)
            ?.scale(scale),
        mesh,
        data: CAR_DATA,
        pickable: true,
        getPosition: d => d,
        getColor: [245, 34, 45],
        wireframe: false,
        updateTriggers: {
          getTransformMatrix: matrix?.vehicleRelativeTransform
        },
})]
```

twojtasz, 2022-06-03:
considering this has been answered, I'm going to close this issue.  Please re-open or file a new issue.

closed 2022-06-03 by twojtasz