aurora-opensource/streetscape.gl #523: Blank Screen in Get-Started example

YashKhorja, 2022-01-05:
I have followed the instructions and started with example:get-started but I am getting screen blank. I am getting below error
[image]

According to the error I am getting, the problem is with the methods using with [XVIZFileLoader](https://github.com/uber/streetscape.gl/blob/master/examples/get-started/src/log-from-file.js#:~:text=export%20default%20new-,XVIZFileLoader,-(%7B), i.e. `on` and `connect`. It is not considering the methods of `XVIZFileLoader`.

SlahserZ, 2022-01-10:
seem like data error.

```
onXVIZMessage = message => {
    switch (message.type) {
      case LOG_STREAM_MESSAGE.METADATA:
        this._onXVIZMetadata(message);
        this.emit('ready', message);
        break;

      case LOG_STREAM_MESSAGE.TIMESLICE:
        this._onXVIZTimeslice(message);
        this.emit('update', message);
        break;

      case LOG_STREAM_MESSAGE.DONE:
        this.emit('finish', message);
        break;

      default:
        this.emit('error', message);
    }
  };
```

you can find it in xviz-loader-interface,  the `message` is null, it leads this crash.

YashKhorja, 2022-01-10:
Yeah but I have just cloned and started the app and didn't make any changes. So it should be working right?

SlahserZ, 2022-01-11:
> Yeah but I have just cloned and started the app and didn't make any changes. So it should be working right?

I think so, you can try other data bag

lesun90, 2022-02-02:
> Yeah but I have just cloned and started the app and didn't make any changes. So it should be working right?

The problem comes from the new update of loaders.gl package. Follow this fix for the temporary fix.

https://github.com/aurora-opensource/xviz/issues/663

twojtasz, 2022-06-03:
This should be fixed in @xviz/*@1.0.11.

You can see commits and published version commits at
https://github.com/aurora-opensource/xviz/commits/master