aurora-opensource/streetscape.gl #525: [proposal] remove streambuffer in child class.

SlahserZ, 2022-01-14:
I'm rebuilding this project by ts, and found a problem.

in lviz loader, you could use `streambuffer` which be from child class by context switch. but it's so loose. 

so I think we can put the `streambuffer` into lviz-loader and remove `streambuffer` in child class. if you ac this, I will fix it.

twojtasz, 2022-04-14:
what do you mean rebuilding in TS? You are rewriting the whole thing in TS.   I have some ts defs.  would be great to coordinate.

SlahserZ, 2022-04-20:
> what do you mean rebuilding in TS? You are rewriting the whole thing in TS. I have some ts defs. would be great to coordinate.

yes, I have rebuild myself project by typescript, but found @xviz/xxx is heavier than my mind, so sorry, I have given this plan up.

closed 2022-05-09 by SlahserZ