aurora-opensource/streetscape.gl #528: Way to visualize voxel grids/ octrees

Chimera-AI, 2022-01-22:
Dear devs,

Is there a way to add voxel grid / octree support in your viewer?
Currently, I use the polygon primitive to wrap a pointcloud representation of voxel grids by cubes to show a it. Basically my points are centroids of the voxel grid.
However, as you might also figure out, this naive approach gets very slow as the size of pointclouds increase.
A native support for adding more shape options for points might work a voxel volume will just surround the point centers for an octree.
Clients can always convert their original pointclouds into desired voxel grid/ octree and then create a sparse pointcloud with each point representing the centroid.
I feel that for a 3D viewer, that aims to be visualize data for autonomous driving will benefit a lot from a voxel grid / octree viewer.

tim-wojo, 2022-01-23:
This has come up before, but we need a dataset to drive a custom layer first and then we can refine the spec for the data and work on a default implementation.

I'm not sure I follow your point-cloud to voxel use-case, it sounds like your talking about quantizing the PC data into a voxel grid, and then using voxels as a clear representation that it covers not just a point but a volume?

I would be curious on a more detailed proposal.  I'm thinking it would have to account for some parameters to the voxel space such as it's fixed origin (assuming it would be fixed such that we have a stable grid over time), and some parameters to the space (how big each voxel is), such that it could represent various depths.  Those may be obvious, but ideally a proposal and example would explicitly cover such cases.

Chimera-AI, 2022-02-06:
Hi Tim,

You are correct in your understanding. You can quantize your pointclouds into voxel grids for example, The library open3D has utilities to convert pointclouds to grids for example.

The biggest usecase will be to display an occupancy grid/map which is created from a raw point cloud. Algorithm like octomap is one such example...

An octomap will take a dense point cloud (from 3D rgbd sensors) or a conventional pointcloud from LiDARs and will create an occupancy grid. Usual parameters will be resolution ie size of the voxel as well as the world size. There are many 3D datasets available online. Even KITTI has pointclouds tgat can be used to create occupancy grids.

The handling of origin is exactly same as a pointcloud. You can have a fixed origin, or you can have a relative origin moving with your sensor.

Current way of achieving a voxel grid in streetscape is by having a quantized point cloud where every point is a voxel centroid. Then use the polygon objects to create a voxel cube around that. However this process will not scale well if the pointcloud is dense.

An easy solution to this is to just allow points to be represented as cubes natively with some specified dimensions.
Currently, points in streetscape has a radius property. Giving a shape property, or supporting a cube primitive will be relatively easy to have.

The application developers can then just use those primitives to display their output by performing the display calculations on their side and just streaming the final voxel centers. We do exactly that but currently need to stream all the polygons as well which slows the display a lot. We acheive 30 FPS of display performance in streetscape on a normal quantized pointcloud which drops down to 10 FPS when streaming cubes around these points. Representing a voxel grid as one will be much helpful than displaying it using pointclouds as it doesnot help us in separating true holes with the fake ones.

I am on vacation currently, but can soon provide you some visual examples if you need.